rusefi/rusefi_documentation #240: Extremely noisy external MAP

ElDominio, 2022-10-07:
https://rusefi.com/online/view.php?log=1068

You can see MAP being VERY noisy in upper RPMs/MAPs, it is connected to the board directly, will follow up thread with pics

ElDominio, 2022-10-07:
Pics

ElDominio, 2022-10-07:
[image]

[image]

[image]

stefanst, 2022-10-07:
That looks like a 25ms period at 6000 rpm. If it's an 8 cylinder it could just be intake pulses. Some smoothing and you should be OK.

rusefillc, 2022-10-07:
@stefanst that's a 6 cylinder nissan 350z