rusefi/rusefi_documentation #244: Reflect "wiki edit has to be restricted" facts on relevant wiki article

rusefillc, 2022-11-08:
* we shall articulate why editing is restricted see #208

* we shall articulate fact that https://github.com/orgs/rusefi/teams/doc-team exists

rusefillc, 2022-11-08:
@mi-hol can you please help?

mck1117, 2022-11-08:
why does that need specific documentation? why not just have instructions for how to contribute to the wiki?

rusefillc, 2022-11-08:
@mck1117 are you looking to debate or help? Please use your best judgement if you are available to edit any relevant articles/pages.

rusefillc, 2022-11-08:
Kind of relevant https://github.com/rusefi/rusefi_documentation/issues/165 but no idea what action if any is needed. Are we supposed to remember why things are done the way they are done or shall we keep the reasons written down?

mi-hol, 2022-11-13:
> why not just have instructions for how to contribute to the wiki?

From my view this is the right approach but it moves the burden on to reviewers to **only merge appropriate changes**.
Defining review rules would allow for (partial) automation.
As example a rule like:
- enforce 4 eyes principle, author of change and person that merges a change can't be the same?

Suggestions very welcome

rusefillc, 2022-11-13:
I am sorry I am barely engaging for multiple reasons :(

mi-hol, 2022-11-13:
> I am barely engaging for multiple reasons

is this due to "Andrey does not believe in text documentation on wiki"  as mentioned in https://wiki.rusefi.com/Documentation-Strategy/ ?

rusefillc, 2022-11-13:
Yes that's one of the reasons

https://github.com/rusefi/rusefi/issues/4722 for instance is much higher priority

mi-hol, 2022-11-13:
During my career as developer I had to learn the following principle in order to be successful and have time for new delopment.
The principles:

DRY means "Don't Repeat Yourself"
So when a piece of software was deployed its users came back with questions. Usually after a short period of the time the same questions came up again and again.
My strategy to have still time for new development was let users become experts for the piece of software themself by first asking RTFM?

RTFM means "Read The Fine Manual" obviously this required to create a manual during mainly the test phase of development too.

rusefillc, 2022-11-13:
I think we have agree on the principle see https://github.com/rusefi/rusefi/wiki/Contributors-please-work-smart

We are chatting via github which is not effective.

mi-hol, 2023-01-27:
> * we shall articulate fact that https://github.com/orgs/rusefi/teams/doc-team exists

@rusefillc this seems the last part to be documented before issue can be closed.

1. Topic I plan to add are:
- role of doc-team 
- how to join doc-team 
- how to leave doc-team 

Any other topics missing?

2. I tried to get feedback from this team but only you and David ever replied.
Should we remove "inactive" team members?

rusefillc, 2023-01-28:
In my opinion we are too far in the wrong area of "content/progress made" (low) and process/policy/cleanup

As of Jan 2023 rusEFI is at least stagnating if not dead. I am excited to play with my test mules and make some progress in a few areas, I am absolutely not excited about process/policy/cleanup.

mi-hol, 2023-01-28:
the frustration expressed should be discussed in a talk

mi-hol, 2023-01-31:
as explained [here](https://github.com/rusefi/rusefi_documentation/issues/261#issuecomment-1410044048) this issue is blocked by a lacking decision from architect

mi-hol, 2023-03-09:
Because of " I am absolutely not excited about process/policy/cleanup. " I'm removing myself from this topic