rusefi/rusefi_documentation #245: wiki@rusefi search results

rusefillc, 2022-11-08:
https://wiki.rusefi.com/search.html?q=trigger order of results looks problematic to me

Specifically pages with "trigger" in the URL are not as high as I would like those to be - in this case https://wiki.rusefi.com/Technical/All-Supported-Triggers/

[image]

mi-hol, 2022-11-13:
@rusefillc I'd suggest to remove label "wiki location drama" and to create a new label "wiki search optimization"

chuckwagoncomputing, 2022-12-12:
https://github.com/mkdocs/mkdocs/issues/1918

chuckwagoncomputing, 2023-01-07:
https://github.com/mkdocs/mkdocs/pull/2590 is a PR that would also massively improve our search results. I have left them a comment, hopefully they will merge it.

rusefillc, 2023-01-23:
@chuckwagoncomputing they have that PR from 2021 :(

mi-hol, 2023-01-27:
At least searching via Google seems to work better now according to Matt's comment:
"Matthew Kennedy
 I just want to say that I googled a question about rusEFI and actually found the relevant wiki page, which was a first" on https://rusefi.slack.com/archives/CF8NQ1RU6/p1674502586139129

mi-hol, 2023-01-27:
> @chuckwagoncomputing they have that PR from 2021 :(

triggered by Andrey's finding I dug a bit deeper into mkdocs and found [this quite concerning discussion ](https://github.com/mkdocs/mkdocs/discussions/2410) 

On the other hand I found [pandoc](https://pandoc.org/index.html) recently as it converts MD to PDF and various other common document formats.
Output to HTML or Wiki formats is another feature, therefore I wonder if we should consider pandoc as an alternative to mkdocs.
May I ask for your views?

chuckwagoncomputing, 2023-01-27:
Good finding on that discussion.

Pandoc doesn't fit our needs. Markdown converted to HTML does not a usable website make.

I'm thinking about trying out the [material theme](https://github.com/squidfunk/mkdocs-material), which is maintained by someone else. That might fix #269, but I don't think it will do anything for this issue.

chuckwagoncomputing, 2023-01-27:
Actually it seems that there was a follow-up to that discussion [here](https://github.com/mkdocs/mkdocs/discussions/2442) and that @oprypin is now maintainer as of mid-2021. The [field boost issue](https://github.com/mkdocs/mkdocs/issues/1918) is from earlier than that, but [the PR for nav path in search results](https://github.com/mkdocs/mkdocs/pull/2590) was under their maintainership and still hasn't been merged.

chuckwagoncomputing, 2025-11-22:
This is massively improved with #597

closed 2025-11-22 by chuckwagoncomputing