rusefi/rusefi_documentation #246: Bring HOWTO slack up to date

rusefillc, 2022-11-08:
https://github.com/rusefi/rusefi/wiki/HOWTO-join-slack-channel image no longer needed? Some content is no longer needed in light of https://rusefi.com/forum/viewtopic.php?p=46606#p46606

mi-hol, 2022-11-08:
@rusefillc happy to help but unclear on the "what" and "how" because

1. never used Slack before
2. https://github.com/rusefi/rusefi/wiki/HOWTO-join-slack-channel was updated by yourself a few minutes ago.

Hence "what" needs to be updates is unclear ATM ans my telepathic abilities are limited too :)

mi-hol, 2022-11-14:
@rusefillc re "https://github.com/rusefi/rusefi/wiki/HOWTO-join-slack-channel **image no longer needed**?" what exact image do you believe is no longer needed?

rusefillc, 2022-11-14:
https://github.com/rusefi/rusefi/wiki/FAQ/images/HOWTO_join_rusEFI_slack_channel/slack_channel_invite.png seems to show old invite page?

closed 2022-11-18 by rusefillc