rusefi/rusefi_documentation #250: rusEFI book - documentation in linear form

rusefillc, 2022-11-14:
We have four instanced of wiki but @eldominio is confident that a "book", i.e. a linear representation of the same content is something of a high priority

rusefillc, 2022-11-14:
Somewhat related to #168

rusefillc, 2022-11-14:
"the book" does not imply that we move the source of truth into a linear form

I guess for PDF purposes content would have to be copy-pasted say once a year. In order to facilitate the once a year PDF production I guess first we need an amazing index wiki page such that once a year one could go through many pages in the order of the index and copy-paste content into a PDF book.

mi-hol, 2023-01-31:
dependency for creating a book is from my view to finish task 'Identify "orphaned" documents having no references to index / Table of Content' in https://github.com/rusefi/rusefi_documentation/issues/297

mi-hol, 2023-01-31:
> "the book" does not imply that we move the source of truth into a linear form

Referencing all relevant documents in the 'Table of Content' makes it kind of a "linear" story.
A good story is required for a book to make it successful.

mi-hol, 2023-01-31:
> first we need an amazing index wiki page

we are aligned on this task :)

mi-hol, 2023-02-02:
> I guess for PDF purposes content would have to be copy-pasted say once a year. In order to facilitate the once a year PDF production I guess first we need an amazing index wiki page such that once a [...]

the manual copy-paste process can be replaced with automation outlined in #1

rusefillc, 2024-02-20:
https://github.com/rusefi/rusefi/wiki/rusEFI-Quick-Start-PDF