rusefi/rusefi_documentation #255: Trigger-Configuration-Guide.md misses 2 "polarity" pictures in "I'm a Nerd and Want to Know More" section

mi-hol, 2022-11-17:
Here are some oscilloscope screenshots of a different trigger wheel, in this case a 24-1 designed specifically for bench testing

...

First, here's the wrong polarity:

< wrong polarity scope image >

...

And here's the correct polarity

< correct polarity scope image >

mi-hol, 2023-03-19:
@rusefillc https://github.com/rusefi/rusefi_documentation/blob/master/Trigger-Configuration-Guide.md
contains picture links to IMGUR hosting.

To avoid copyright violations when copied, I wonder if you are the copyright owner?
the links are:
[image]
[image]

mi-hol, 2023-03-19:
4 more files with IMGUR links 
[image]

rusefillc, 2023-03-19:
I assume that @mck1117 has made those

mck1117, 2023-03-19:
indeed I did make those, do whatever you want with them

mi-hol, 2023-03-26:
@mck1117 would you perhaps have the images required to complete below ToDos in your IMGUR gallery?
I'd just need a URL to your IMGUR gallery as a starting point so I can search for them

ToDo:

add correct polarity scope image
add wrong polarity scope image

mi-hol, 2023-06-13:
removing myself as my hands are tied without support

chuckwagoncomputing, 2025-12-17:
Deleted in #635