rusefi/rusefi_documentation #257: REO does not display latest tunes

duhafnusa4, 2022-11-21:
I was looking for a specific tune that a friend said they uploaded, it was recent so I expected to see it on the main page of reo. Scrolling down I see an upload of his from 10-15-2022. 

[image]

If I click on his username, then click on his project name taking me to all his stuff on REO, I can now see a later upload from 11-3-2022

[image]

1) Why can I not see the latest tunes uploaded on the REO home page?
2) What is the order that the tunes shown on the homepage are listed in?

duhafnusa4, 2022-11-28:
bump @rusefillc

mi-hol, 2022-12-11:
@rusefillc it seems this issue is in the wrong repository. 
Shall I move it to rusefi/web_backend via "transfer issue"?