rusefi/rusefi_documentation #259: Hide/remove all references to discovery/frankenso

rusefillc, 2022-12-03:
These days we target people who have purchased fully assembled hardware

Too much of documentation referencing discovery board or Frankenso is pretty much obsolete now. While we should not _remove_ it we shall definitely _hide_ it or _mark_ it.

rusefillc, 2022-12-03:
@mi-hol would you be available and interested to locate such pages? just having a list would be a great help towards deciding how we hide it

mi-hol, 2022-12-05:
> locate such pages?

doesn't GH search like https://github.com/rusefi/rusefi_documentation/search?q=Frankenso produce the list of files?

mi-hol, 2022-12-05:
to make this an actionable task a clear definition of "what should be done" is needed. WWhile we should not remove it we shall definitely hide it or mark it." is unfortunately **not** actionable

rusefillc, 2023-01-28:
Do not see actionable items or specific questions.

closed 2023-01-28 by rusefillc