rusefi/rusefi_documentation #261: update of rusefi_documentation/Dev/Dev-Status.md required

mi-hol, 2022-12-06:
@rusefillc "Current" status seems to be of about one year ago.

I see it as a great "marketing" document.
Assuming all relevant issues have been link per feature I could update status icons, but a thorough review will be required by technical expert(s) **before** it should be merged.

rusefillc, 2022-12-11:
@mck1117 @pk386 @chuckwagoncomputing second round?

chuckwagoncomputing, 2022-12-11:
It seems like a nice page to have - if it's actually referred to.

rusefillc, 2022-12-11:
it is definitely abandoned. I've forgotten about it for sure, and it has out of date info!

rusefillc, 2022-12-11:
https://github.com/rusefi/rusefi/blob/master/misc/work_in_progress.md is another equally abandoned "status" file.

mi-hol, 2022-12-11:
> It seems like a nice page to have - if it's actually referred to.

Very useful for potential buyers, but if it's not reflecting current capabilities, a lost sales opportunity is the result.

rusefillc, 2022-12-11:
This is related to #111

We have two "in progress" documents and three(!!!) sources of current feature set

mi-hol, 2022-12-11:
> We have two "in progress" documents and three(!!!) sources of current feature set

good catch, please add URL links for feature set.
This will allow to understand the issue and to create a plan to fix it

mi-hol, 2022-12-11:
@rusefi/doc-team 
To facilitate planning next milestone for documentation, I created a new label "feedback needed" and assigned to issues with unclear status or long due activities.
As the next step I'll try to assign it to the correct owner for resolution.
Please provide feedback in case I got this wrong

mi-hol, 2022-12-11:
@rusefi/doc-team 
first progress:  

- 11of 63  issues have been closed.

next step: 

- prioritize remaining 52 issues.

My proposal is to work on "bugs" first.

Would love to read comments or concerns :)

rusefillc, 2023-01-28:
somehow this has started with a desire for plans and switched into asking me to edit content. I am confused and overwhelmed, my default response is to run away

mi-hol, 2023-01-31:
> somehow this has started with a desire for plans and switched into asking me to edit content.

with a single point of control, decisions that are not taken, block everyone on a team that would like to contribute from making progress.

> I am confused and overwhelmed, my default response is to run away

Will not cure the problem.
The problem is a typical project management problem encountered already in early 1970s at IBM.
"Complex programming projects cannot be perfectly partitioned into discrete tasks that can be worked on without **communication** between the workers and without establishing a set of complex interrelationships between tasks and the workers performing them."

It triggered the creation of THE BOOK for project management "The Mythical Man Month"
https://en.wikipedia.org/wiki/The_Mythical_Man-Month#Communication
"Communication
To avoid disaster, all the teams working on a project should remain in contact with each other in as many ways as possible (e-mail, phone, meetings, memos, etc.). Instead of assuming something, **implementers should ask the architect(s) to clarify their intent on a feature they are implementing, before proceeding with an assumption** that might very well be completely incorrect. The architect(s) are responsible for formulating a group picture of the project and communicating it to others."

rusefillc, 2023-01-31:
For last 24 months 90% of the changes are done by two contributors - @mck1117 and myself. @mck1117 and myself are totally NOT communicating. Once we remove @mck1117 as non-communicating entity that leaves myself. As of today my plans for this project are unclear.

Many things are just consequences of the above statement.

@pk386 FYI.

rusefillc, 2023-01-31:
@chuckwagoncomputing see drama above - it reminds me of https://github.com/rusefi/rusefi/pull/5019

my problem with https://github.com/rusefi/rusefi/pull/5019 is I do not major value, I definitely see at least minor damage. Inability to make the decision fast takes me to the uncomfortable for me zone of "this needs proper attention but no motivation to invest major decision making effort into what seems like something not important"

mi-hol, 2023-01-31:
> For last 24 months 90% of the changes are done by two contributors

at least for doc repo there are now others too :)
[image]

mi-hol, 2023-06-17:
Most former contributors seem to have moved to https://github.com/FOME-Tech/wiki
closing therefore.

closed 2023-06-17 by mi-hol