rusefi/rusefi_documentation #262: wiki references wrong download URL

mi-hol, 2022-12-11:
https://wiki.rusefi.com/Home/  shows
[image]
Image uses URL https://wiki.rusefi.com/Home/Download
but this URL has no content to show/download
[image]

next menu item "Download" uses the correct URL https://wiki.rusefi.com/Download/
[image]

rusefillc, 2022-12-12:
I am very confused see #267

rusefillc, 2022-12-14:
I open in Chrome
I mouse over image with the box

I see status promising https://wiki.rusefi.com/Download/

I click on the image and I get https://wiki.rusefi.com/Download/

[image]

Does this contradict this issue statement or do I get it wrong or was it already fixed?

mi-hol, 2022-12-14:
> Does this contradict this issue statement

yes and no because I used a different URL, where the link is still wrong
[image]
Could you please verify using https://wiki.rusefi.com/Home/
This URL is used after navigation from left sidebar is used

PR #298 (2022-12-19): remove home link from mkdocs sidebar
This will fix #262. We don't need a home link in the sidebar in the first place, as there is already one at the top. The reason that this link causes an issue is that it points at /Home, while the configured canonical URL for the home page is at /, so the links generated in the home page assume that the current URL is /. If the URL is in fact /Home, all the links in the home page no longer work.
closed 2022-12-19 by rusefillc