rusefi/rusefi_documentation #266: Add link to wiki for downloading history versions

mi-hol, 2022-12-11:
Downloads are currently limited to 2 versions:
[image]

add linh allowing for downloading history versions (i.e. for troubleshooting)

rusefillc, 2022-12-12:
which exact page is affected? I see

[image]

mi-hol, 2022-12-15:
https://github.com/rusefi/rusefi/releases?page=6 has oldest release [2017.03.31_release_1.0.2](https://github.com/rusefi/rusefi/releases/tag/2017.03.31_release_1.0.2)
SF has <html>
<body>

rusefi.r46.zip | 2013-10-20
-- | --

</body>
</html>[rusefi.r46.zip](https://sourceforge.net/projects/rusefi/files/rusefi.r46.zip/download)	2013-10-20

closed 2022-12-15 by mi-hol