rusefi/rusefi_documentation #267: https://github.com/rusefi/rusefi/wiki sidebar Downloads link is broken

rusefillc, 2022-12-12:
https://github.com/rusefi/rusefi/wiki sidebar Downloads link is broken?! same link on main page with image not broken? how come sidebar Download reference is handled differently and points to https://github.com/rusefi/rusefi/Download without "wiki" in URL?!

huh?! how comes
```
- [Documentation Strategy](Documentation-Strategy)

```
works while
```
### [Downloads](Download)
``` is broken?!

@chuckwagoncomputing

chuckwagoncomputing, 2022-12-12:
This is the reverse issue of #177, which has begun to work, but this has not. Wiki2 and Wiki3 sidebars are in separate files, so the Wiki2 sidebar link could be made a full link to work around this issue.

rusefillc, 2022-12-12:
@chuckwagoncomputing do you have any idea why is that an issue in the first place? I am totally fine making it the full link, but WHY?

mi-hol, 2022-12-12:
From my view the real issue we should focus on resolving first is:
document how relative links are transformed to absolute links in:
- wiki3 (source: https://github.com/rusefi/rusefi_documentation target: https://wiki.rusefi.com)
- wiki2 (source: https://github.com/rusefi/rusefi/wiki target: https://github.com/rusefi/rusefi/wiki)
- wiki2 to wiki3 automated sync (this is where I suspect the root cause of error!)

This documentation and the resulting configurations are usually kept in a private repository with very limited number of users.
I'd describe this as "project web infra structure"  topic  and already created related #249 and #264 "sub topics"

Does this make sense?

chuckwagoncomputing, 2022-12-13:
@rusefillc As far as I can tell, this is an issue on Github's end.

@mi-hol [https://github.com/rusefi/rusefi_documentation](https://github.com/rusefi/rusefi_documentation) is the source for both wiki2 and wiki3. How relative links are transformed for wiki2 is handled by Github. We have no say over how it works, and it can and has changed without notice. How relative links are transformed by MkDocs for wiki3 in documented [by MkDocs](https://www.mkdocs.org/user-guide/writing-your-docs/).

PR #299 (2022-12-19): use full link for Downloads on wiki2
This should fix #267.
closed 2022-12-19 by rusefillc