rusefi/rusefi_documentation #268: [Proposal] make "master" a protected branch to enforce update via PRs only

mi-hol, 2022-12-13:
I'd recommend to enforce certain workflows for "master" branch via "protection rules"- Possible "protection rules" such as requiring an approving review or passing status checks for all pull requests merged into the protected branch.
How to is documented in (https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/defining-the-mergeability-of-pull-requests/managing-a-branch-protection-rule)

Background:
every member of doc-team has "merge" permission and can push to "master" branch unintentionally.

mi-hol, 2023-01-03:
this change links nicely with #244

mi-hol, 2023-01-12:
Andrey is fine with recommendation provided by doc-team

chuckwagoncomputing, 2023-01-13:
I've already spoken to @mi-hol about this, but just so everyone is on the same page:
This will also mean disabling editing from https://github.com/rusefi/rusefi/wiki

rusefillc, 2023-01-13:
[image]

rusefillc, 2023-01-13:
even with branch protection I can still make direct commits https://github.com/rusefi/rusefi_documentation/commit/a895791d0c4641f57a5641d7ad5bfd3ad56e603a

to be honest that works for me

rusefillc, 2023-01-13:
> I've already spoken to @mi-hol about this, but just so everyone is on the same page: This will also mean disabling editing from https://github.com/rusefi/rusefi/wiki

at the moment exactly one person can edit https://github.com/rusefi/rusefi/wiki directly and that's me

please clarify the thought process here, what are we trying to solve how?

chuckwagoncomputing, 2023-01-13:
As I understand it, the goal is to keep people from making bad commits that someone has to fix later.

Just to confirm, have you also selected "Require status checks to pass before merging"?

rusefillc, 2023-01-13:
I did now

[image]

chuckwagoncomputing, 2023-01-13:
OK Thanks.

closed 2023-01-13 by chuckwagoncomputing

mi-hol, 2023-01-15:
Thank you all, from my view this is a great process improvement!