rusefi/rusefi_documentation #289: Relative URLs are broken

nmschulte, 2022-12-16:
First, should https://github.com/rusefi/rusefi/wiki/ work correctly along with https://wiki.rusefi.com/?  Does https://wiki.rusefi.com/ need generated after changes are made?

---

As an example, the _All Supported Triggers_ page is broken, both on github.com and rusefi.com, but in different ways:

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers

- e.g. the _Triggers_ link is broken, linking to https://github.com/Triggers/Trigger
- none of the images display as they also reference relative to https://github.com/

https://wiki.rusefi.com/ (possibly this is outdated version? unsure how this works)

- e.g. the _Triggers_ link is broken, linking to https://wiki.rusefi.com/Technical/Triggers/Trigger
- the images _do_ display (possibly because this is an outdated version?)

mi-hol, 2022-12-16:
> https://wiki.rusefi.com/ (possibly this is outdated version? unsure how this works)

https://wiki.rusefi.com/ is the currently active version and https://github.com/rusefi/rusefi/wiki should be identical

Details in https://github.com/rusefi/rusefi/wiki/HOWTO-contribute-to-documentation#technical-implementation-details

I'll check https://wiki.rusefi.com/Technical/Triggers/Trigger 
Please add reference errors here, should you find more.

mi-hol, 2022-12-16:
Weird incorrect output from markdown-link-validator for 
```
C:\Users\mholzer\Documents\git\rusefi_documentation\Triggers\Trigger.md

Found 0 links:
    0 valid
    0 invalid
```
while VSCode found many link errors. I've fixed them and merged a PR. 
Will have to dig deeper :(

nmschulte, 2022-12-16:
> Please add reference errors here, should you find more.

These are all of the ones I found:
```
 Dev/Dev-Status.md                                   | 188 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++------------------------------------------------------------------------------------------
 HOWTO/FSIO.md                                       |  12 ++++++------
 Hardware/connectors/OEM-121-pin-connectors.md       |  26 ++++++++++++-------------
 Hardware/connectors/OEM-connectors.md               |  40 +++++++++++++++++++-------------------
 OEM-Docs/Honda/2003-element/digital/readme.md       |   2 +-
 Overview/TS_generated/rusEFI-project-Advanced.md    |  34 ++++++++++++++++-----------------
 Overview/TS_generated/rusEFI-project-Base-Engine.md |  20 +++++++++----------
 Overview/TS_generated/rusEFI-project-Controller.md  |  46 ++++++++++++++++++++++----------------------
 Overview/TS_generated/rusEFI-project-Cranking.md    |   8 ++++----
 Overview/TS_generated/rusEFI-project-Fuel.md        |  24 +++++++++++------------
 Overview/TS_generated/rusEFI-project-Idle.md        |  10 +++++-----
 Overview/TS_generated/rusEFI-project-Ignition.md    |  12 ++++++------
 Overview/TS_generated/rusEFI-project-Sensors.md     |  38 ++++++++++++++++++------------------
 Overview/TS_generated/rusEFI-project.md             |  16 ++++++++--------
 Technical/All-Supported-Triggers.md                 | 106 ++++++++++++++++++++++++++++++++++++++++++++++++++---------------------------------------------------
 wiki-tools/HOWTO-validate-links.md                  |   4 ++--
 16 files changed, 293 insertions(+), 293 deletions(-)
 ```
 
 Also, `_Sidebar.md`'s _Home_ link.

nmschulte, 2022-12-16:
re: #290, there's still issues about the GitHub hosted wiki:
- https://github.com/rusefi/rusefi/wiki/Trigger's "See list of all supported triggers" link at top links to https://github.com/Technical/All-Supported-Triggers.md
- images reference relative to https://github.com/ vs https://github.com/rusefi/rusefi/wiki/ (e.g. https://github.com/Images/triggerinfo.png) and so obviously do not display (as they 404)
[image]

My initial post was about the link to _Triggers_ from `All-Supported-Triggers.md` (its "See also Trigger" link at top), but it's the same issue across the board.
[image]

mi-hol, 2022-12-16:
> These are all of the ones I found

Could you please elaborate how you found them?

nmschulte, 2022-12-16:
I noticed the issue was with URLS with leading `/`, and so found them via: `git grep -Il '(/.*)' -- **/*.md`  (note that's capital `I`, lowercase `l`).

mi-hol, 2022-12-16:
> re: #290, there's still issues about the GitHub hosted wiki:
> 
> * [https://github.com/rusefi/rusefi/wiki/Trigger's](https://github.com/rusefi/rusefi/wiki/Trigger's)

Could we focus on wiki3  [https://wiki.rusefi.com](https://wiki.rusefi.com/) **for now**, because there seem to be long standing issues with automation to sync doc repository  to wiki2  https://github.com/rusefi/rusefi/wiki/  (see https://github.com/rusefi/rusefi_documentation/labels/wiki%20location%20%26%20process%20change)

> I noticed the issue was with URLs with leading `/`

I've actually inserted leading `/` intentionally to allow for invalid link detection via VS Code & markdown-link-validator!
Sadly there is currently no process to verify a change to wiki before it's merged (see #249)

My conclusion from above is that automation to sync doc repository  to wiki2 & wiki2 is likely the root cause of issues found and will investigate further.

nmschulte, 2022-12-17:
FYI also this looks strange:
[image]

chuckwagoncomputing, 2022-12-18:
Am I understanding correctly that the issue here was that @mi-hol added leading slashes that don't work, and this can be closed?

nmschulte, 2022-12-18:
> Am I understanding correctly that the issue here was that @mi-hol added leading slashes that don't work, and this can be closed?

Yes I'd agree with that: there are enough issues about the situation regardless, closing one can't hurt.

closed 2022-12-18 by nmschulte