rusefi/rusefi_documentation #291: MAJOR BUG trigger page is broken

rusefillc, 2022-12-16:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers is one of the most important rusEFI documentation pages

At the moment it does not display images
I believe it was displyaing images a few fays ago

ALL ACTIVITY HAS TO HALT UNTIL WE INVESTIGATE HOW DID WE ALLOW https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers to break

chuckwagoncomputing, 2022-12-16:
This is more Github trailing slash buggery that we have no power over. Let's move to a wiki that we do have power over.

mi-hol, 2022-12-16:
> HOW DID WE ALLOW https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers to break

from my view the root cause lies in /rusefi_documentation/.github/workflows/sync-wiki.yaml
```
  - name: Generate static pages using mkdocs
      run: |
        shopt -s extglob
        cd mkdocs
        mkdir docs
        cp -r ../!(mkdocs) docs
        mkdocs build
```

- seems to use a "default" configuration of mkdocs
- misses to check for result of build 
- no tests to verify correctness of key files are performed
- but the next step "Upload static results to wiki3" publishes result directly

rusefillc, 2022-12-16:
I do not follow

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers is source, not result of mkdocs process

rusefillc, 2022-12-16:
While it's source it's also PRIMARY instance

I.e. we stumble over ourselves in terms of how to implement the transition

chuckwagoncomputing, 2022-12-16:
> * seems to use a "default" configuration of mkdocs

incorrect, configuration is [here](https://github.com/rusefi/rusefi_documentation/blob/master/mkdocs/mkdocs.yaml)

> * misses to check for result of build
> 
> * no tests to verify correctness of key files are performed
> 
> * but the next step "Upload static results to wiki3" publishes result directly

Yep, that's some very poor engineering on my part.

---
HOWEVER - MkDocs has nothing to with the issue at hand.

mi-hol, 2022-12-16:
> I do not follow
>  https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers is source,

I have edited /rusefi/rusefi_documentation/Technical/All-Supported-Triggers.md to fix link issues (see PR #273 related to 
#123)  
"source" implies this should not have been done.
If I understand this correctly, the only safe way to prevent such an unintentional "error" would be:
- the file should not exist at all 

**OR** 

-  be protected from edits.

rusefillc, 2022-12-17:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers is the URL which I would prefer to absolutely work.

https://github.com/rusefi/rusefi_documentation/blob/master/Technical/All-Supported-Triggers.md is an exact clone on the file which would be displayed on https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers

Meaning that any edits to https://github.com/rusefi/rusefi_documentation/blob/master/Technical/All-Supported-Triggers.md have to account for the fact that it's exactly the file which has to be working properly at https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers

Good news we have #293

chuckwagoncomputing, 2023-01-01:
I opened a ticket with Github 9 days ago re. trailing slashes, and have received no response. I think Github doesn't care.

rusefillc, 2023-01-01:
@chuckwagoncomputing ticket URL please?

anyway we've fixed triggers catalog so closing this ticket

closed 2023-01-01 by rusefillc

chuckwagoncomputing, 2023-01-01:
@rusefillc It's a non-public Github support ticket.