rusefi/rusefi_documentation #293: Flat .md structure as a way to reduce migration pain/consistency

rusefillc, 2022-12-17:
For instance https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers is made out of https://github.com/rusefi/rusefi_documentation/blob/master/Technical/All-Supported-Triggers.md - note how "Technical" disappears

Looks like to survive all the three wikis mess we would have to move all .md folders into root folder

This seems to be at least part of the of #291 maybe #292 else

mi-hol, 2022-12-17:
Sorry but this approach suits only as a temporary workaround for one single file because document editor's experience is still broken!
**We need a solution that works for all involved parties.**

[image]

chuckwagoncomputing, 2022-12-17:
I hate flat directory structures, but it appears this is the only way to make Github wiki actually work.

@mi-hol Unfortunately whatever tool you are using there doesn't understand how Github wiki works, and that's not our fault. Maybe you should open a feature request with whoever has made it?
Links and images that actually work are a high priority.

rusefillc, 2022-12-19:
just moved 129 .md files manually there is a good chance I've missed some

Need help locating what have I missed? Might be a one line bash?

chuckwagoncomputing, 2022-12-19:
From memory:
`find */* -name '*.md' -exec mv '{}' . \;`
Not near a computer so I'm not sure if this is right, but it's close.

chuckwagoncomputing, 2022-12-19:
Looks like that command works, but beware of name collisions - there are three files named readme.md. Rename them before running the command.
Also, you can run `find . -type d -empty` to find directories that are left empty.

rusefillc, 2022-12-19:
Collisions commit was cherry picked into master https://github.com/rusefi/rusefi_documentation/commit/c425e81bef7e840aa00375b92a574bc91b10d1b1

rusefillc, 2022-12-19:
thank you @chuckwagoncomputing all moved now, we are flat with .md, closing

commit 954a865a (2022-12-19): Flat .md structure as a way to reduce migration pain/consistency fix #293
closed 2022-12-19 by rusefillc