rusefi/rusefi_documentation #296: Slack invitation mechanism

rusefillc, 2022-12-19:
For a couple of years we used externally hosted free application to manage slack invitations - that service is no longer free

Slack build-in "invite link" expires every 30 days LAME

Would be nice to have a free and permanent mechanism for new people to join slack

chuckwagoncomputing, 2023-01-20:
https://github.com/outsideris/slack-invite-automation/issues/151

Apparently the free version of Slack doesn't allow admin-level API access.
Is that actually required?
I don't see anything on https://communityinviter.com that mentions needing paid Slack.

rusefillc, 2023-01-20:
Is what required? Do we want to stop advertising slack and stop inviting to slack?

[image]

https://join.slack.com/t/rusefi/shared_invite/zt-1lua6xsva-n~3TvYnT0Brzak35vQBYQw

[image]

chuckwagoncomputing, 2023-01-20:
I meant:
Would a service really require admin-level access in order to invite people to Slack?

closed 2025-08-10 by rusefillc