rusefi/rusefi_documentation #297: Requirements for documentation and it's change process

mi-hol, 2022-12-19:
- Support an editor to assess correctness of links in a document currently edited (i.e. via native editor feature or  VS Code extensions)
- Allow bulk validation of links in all documents
- Identify "orphaned" documents having no references to it
- links from rusEFI forum to documents should work indefinitely

mi-hol, 2022-12-31:
consider learnings from discussion in PR https://github.com/rusefi/rusefi_documentation/pull/294#issuecomment-1356669551

chuckwagoncomputing, 2023-01-06:
- Support an editor to assess correctness of links in a document currently edited (i.e. via native editor feature or  VS Code extensions)
- Allow bulk validation of links in all documents
   We have enabled strict mode, have a link checking script, and have enabled linting for pull requests (once #360 is merged)

- Allow for structuring content via directories
   WONTFIX

- Identify "orphaned" documents having no references to index / Table of Content
   * already reported by `mkdocs build` but not visible as output is not brought back from gh action
   * example:
      ```
      INFO     -  The following pages exist in the docs directory, but are not included in the "nav" configuration:
      - 1995-3000gt.md
      - 1997-e320.md
      ```
    (for their names see) [427 orphaned md files as of 30Dec2022.txt](https://github.com/rusefi/rusefi_documentation/files/10325621/427.orphaned.md.files.as.of.30Dec2022.txt)

- links from rusEFI forum to documents should work indefinitely
   Do you have a proposal for how to do this magic?

- updates of documents are only possible via Pull Requests to allow for revert
   We can revert without Pull Requests - though pull requests are still a good idea. @rusefillc likes to push to master. If he wants to no longer make that possible for himself, he may do so.

mi-hol, 2023-05-29:
not feasible

closed 2023-05-29 by mi-hol