rusefi/rusefi_documentation #302: Most or all of user-facing pages should not even be mentioning the console text commands

rusefillc, 2022-12-23:
Like https://github.com/rusefi/rusefi/wiki/Cranking

mi-hol, 2022-12-25:
Would you mind to mention the reasoning for this change?

In the example above I see no benefit as from the visual impression only "clutter" (actually distracting users), gets added via "More"
[image]

expands to:
[image]

rusefillc, 2022-12-25:
Commands create wrong impressions that one has to be a software developer to operate the unit, we do not want that impression.

Commands are clearly noise for 99.9999% of the target audience

mi-hol, 2022-12-25:
Now with a user base of 0.0001%, would the better approach be to remove the "noise" altogether?

closed 2023-01-02 by mi-hol