rusefi/rusefi_documentation #305: Cranking: TunerStudio and documentation out-of-sync

mi-hol, 2022-12-24:
[image]

Requires a technical expert to revise content

mi-hol, 2022-12-25:
note: there is a major difference on how the .md is rendered on https://github.com/rusefi/rusefi/wiki/Cranking vs https://wiki.rusefi.com/Cranking/

[image]

chuckwagoncomputing, 2022-12-27:
There's nothing we can do about the gaps in the images on wiki2. And the only way to make the text flow around it as far as I can tell is to put some in a table like it was before. I thought it was less confusing this way, but if others disagree, I can change it back.

mi-hol, 2022-12-27:
> There's nothing we can do about the gaps in the images on wiki2. 

This is more of an argument for killing wiki2

> I can change it back.

There are many issues on this page anyways (please see in first comment) and #313 
Do you use an editor that supports markdownlint?

chuckwagoncomputing, 2022-12-27:
> There are many issues on this page anyways (please see in first comment) and #313

I was responding to the things I know about.

> Do you use an editor that supports markdownlint?

No

nmschulte, 2022-12-28:
> There's nothing we can do about the gaps in the images on wiki2.

If we can set `style="line-height: 0"` or `style="font-size: 0"` on the `<td>` wrapping the image-links, the gaps should go away.

mi-hol, 2022-12-28:
> > There are many issues on this page anyways (please see in first comment) and #313
> 
> I was responding to the things I know about.
> 
> > Do you use an editor that supports markdownlint?
> 
> No

May I suggest a trial with VSCode to see the hugh number of issues?
Setup is described in https://github.com/rusefi/rusefi_documentation/blob/master/HOWTO-validate-links.md#using-vs-code-for-markdown-files

chuckwagoncomputing, 2022-12-28:
> If we can set `style="line-height: 0"` or `style="font-size: 0"` on the `<td>` wrapping the image-links, the gaps should go away.

Github strips most parameters, including style, from HTML tags.

chuckwagoncomputing, 2022-12-28:
> > > Do you use an editor that supports markdownlint?
> > 
> > 
> > No
> 
> May I suggest a trial with VSCode to see the hugh number of issues? Setup is described in https://github.com/rusefi/rusefi_documentation/blob/master/HOWTO-validate-links.md#using-vs-code-for-markdown-files

I'm hesitant to use a tool that doesn't understand the nuances of what we are doing. It has already caused problems by telling you to add leading slashes which did not work.

mi-hol, 2022-12-28:
> It has already caused problems by telling you to add leading slashes which did not work.

that is incorrect understanding!  MarkdownLint in VSCode will only report the "link not found" type of problem but won't fix it.
The user has to make the fix and it turned out my approach (used in previous projects successfully) was wrong for rusEFI.

My learning was to create https://github.com/rusefi/rusefi_documentation/issues/297

My conclusion so far: 
We need tooling (i.e. similar to unit tests in firmware development) to find such issues and prevent them from being merged!

I'd love to see input & feedback in above linked issue