rusefi/rusefi_documentation #308: markdownlint errors need fixing

mi-hol, 2022-12-24:
Validate Markdown Files With MarkdownLint according to [this ](https://matthewsetter.com/tools-that-make-technical-writing-easier-markdown-linter/)blog post revealed over 7K errors.
To avoid follow-on errors in other tools I'd suggest to fix them. 

Luckily most errors can be fixed automatically.

As a follow-up we should activate this check on each commit.
```
PS git\rusefi_documentation> markdownlint-cli2 "**/*.md" "#node_modules" 2>.\wiki-tools\mdl-20221224.log
markdownlint-cli2 v0.5.1 (markdownlint v0.26.2)
Finding: **/*.md !node_modules
Linting: 425 file(s)
Summary: 7557 error(s)
```
[mdl-20221224.log](https://github.com/rusefi/rusefi_documentation/files/10299257/mdl-20221224.log)

mi-hol, 2023-01-02:
after PR #334 
markdownlint-cli2-config v0.5.1 (markdownlint v0.26.2)
Finding: **/*.md !node_modules
Linting: 424 file(s)
Summary: 4072 error(s)

chuckwagoncomputing, 2023-01-06:
I've been working on this today. I've completed files starting with A-H.

mi-hol, 2023-01-08:
@rusefillc sorry this is NOT closed
Again a case for #363

chuckwagoncomputing, 2023-01-08:
You can't just re-open an issue without explaining why.

mi-hol, 2023-01-08:
markdownlint-cli2 v0.6.0 (markdownlint v0.27.0)
Finding: **/*.md !node_modules
Linting: 426 file(s)
Summary: 3510 error(s)

mi-hol, 2023-01-08:
@chuckwagoncomputing 
I must say that I find this a rather crappy approach!
[image]

mi-hol, 2023-01-08:
to avoid misunderstanding I wanted to clarify:
I'm fine to provide a reason, the "crappy approach" is related to "back and forth" as explained further in #363 
ANY issue should be fixed completely, ofter multiple iterations are required to achieve that, because of different view points contributors have!

chuckwagoncomputing, 2023-01-08:
Please state your view as to why this issue should not be closed.

mi-hol, 2023-01-12:
"Documentation as Code" is a concept to apply CI to documentation in order to avoid common errors.
As a "kind of compiler" fpr md files several tools need to validate the content of a changed file and reject it for merging in case of detected errors & warning.

example:
[image]

mi-hol, 2023-01-12:
Current implementation validates **all markdown files** but without any consequence in case of detected error

chuckwagoncomputing, 2023-01-13:
Fixed. Details in #268

mi-hol, 2023-01-15:
@chuckwagoncomputing does the current implementation prevent references with .md extension?
To my current understand this type of error was corrected manually and could re-occur any time unintentionally.

BTW: I Prefix this issue as "EPIC" (as suggested by Andrey) to remain in control of status

chuckwagoncomputing, 2023-01-15:
No it doesn't. An idea I have for how to do that is to modify brokenlinks.sh to accept a flag to make it non-interactive, and just check links.

mi-hol, 2023-01-15:
> modify brokenlinks.sh to accept a flag to make it non-interactive, and just check links.

  not sure if this is the best approach because in my tests [here ](https://wiki.rusefi.com/HOWTO-validate-links/) brokenlinks.sh was extremely slow compared to  markdown-link-validator.
As I understand it currently  markdown-link-validator is actually based on ESLint and consists of [just 49 lines of TypeScript code](https://github.com/webhintio/markdown-link-validator/blob/main/src/bin/markdown-link-validator.ts),
Do you happen to know TypeScript?

Another approach could be "[custom rule](https://github.com/DavidAnson/markdownlint/blob/main/doc/CustomRules.md)" for markdownlint.

chuckwagoncomputing, 2023-01-16:
brokenlinks.sh takes less than a minute on my ~5 year old laptop. It could also be modified to only check the files that were changed in the PR.

If you really want to customize other tools to fit our needs, please feel free to do so. I only recommend not doing so because I don't feel that it's a good use of developer time.

chuckwagoncomputing, 2023-01-17:
The same workflow would trigger whether a file was modified or files were renamed or deleted.
You could just check in the 'run' section, using some git command probably, if any files were renamed or deleted, and act accordingly.

Yes, I was wrong. I thought that the board was skipped if the id was missing, turns out the script says:
`# todo fail once all pages are compliant    exit 1`

mi-hol, 2023-03-09:
close as script now fails if errors are detected

PR #360 (2023-01-07): Lint
Fix #308
Add check on pull requests
closed 2023-03-09 by mi-hol