rusefi/rusefi_documentation #313: embedded HTML code creates issues for MkDocs and results in wrong display on wiki.rusefi.com

mi-hol, 2022-12-27:
## example:

[image]

## root cause:
[image]

## fix:
**use only markdown tags and remove embedded HTML code**

[image]

## affected markdown files
TBD

chuckwagoncomputing, 2022-12-27:
details and summary tags work fine in Wiki3, but you have to add markdown parameters - see [Cranking.md](https://github.com/rusefi/rusefi_documentation/blob/master/Cranking.md) for example.

mi-hol, 2022-12-27:
> details and summary tags work fine in Wiki3, but you have to add markdown parameters

This is really thin ice and from my view won't work reliably without proper documentation and tooling!
As we already see the negative impacts of not having that, a plan to get this back under control would be a starting point.
Anything to contribute?

mi-hol, 2022-12-29:
documentation and tooling is missing, #318 only fixed "tip of iceberg"

chuckwagoncomputing, 2023-01-01:
Documentation now exists in https://wiki.rusefi.com/HOWTO-contribute-to-documentation/
Do you have any ideas for checking this with tooling? Write our own linter? In my opinion, the effort required to write a linter is so much more than the effort for editors to check manually that their edits haven't created a broken details section, that it's not worth writing a linter.

mi-hol, 2023-01-02:
> Do you have any ideas for checking this with tooling?

Do you happen to know https://github.com/DavidAnson/markdownlint-cli2?
Blog post https://dlaa.me/blog/post/markdownlintcli2 makes me optimistic it will fit with our requirements.

my decision factors are driven by benefits & reliability not only by efforts.

> editors to check manually

is unfortunately not an option to trust on because an editor with the only the best intent (like me) may introduce errors unintentionally due to lacking context, skill or requirements for a specific aspect.
What I learned is that there is never stability regarding contributors in a project team!
Meaning someone is always below the required skill level to work "error free".
Tooling is the only option to safeguard against this thread for a project.

chuckwagoncomputing, 2023-01-04:
markdownlint doesn't come anywhere close to doing what we need without writing our own checks for it. I guess if somebody wants to do that, they can, but it seems to me like a poor use of developer time.

mi-hol, 2023-01-05:
> markdownlint doesn't come anywhere close to doing what we need

Could you perhaps add what you know to be needed?

mi-hol, 2023-01-05:
> markdownlint doesn't come anywhere close to doing what we need

did you check https://github.com/DavidAnson/markdownlint-cli2 as the tool I currently use?

chuckwagoncomputing, 2023-01-05:
markdownlint only checks for presence of HTML.
We need HTML, with particular parameters for tags if there is markdown within them, and empty lines after particular closing tags. markdownlint doesn't know how to do any of that.

chuckwagoncomputing, 2023-01-05:
If you want to modify markdownlint to do that, you may. If not, let's close this issue.

mi-hol, 2023-01-08:
again #363 :(
This is my issue to deal with!
[image]

mi-hol, 2023-01-08:
below approach is not what I envisaged

[image]

chuckwagoncomputing, 2023-01-13:
Fixed. Details in #268

PR #318 (2022-12-29): Fix collapsible details
Fixes #313
closed 2023-01-13 by chuckwagoncomputing