rusefi/rusefi_documentation #315: Security issue reporting

mi-hol, 2022-12-28:
in GitHub's Security menue "Security policy • Disabled
Suggest how users should report security vulnerabilities for this repository"

this requires usually a dedicated email address to report potential security issues.

SiteChecker has the following security related findings:

[image]

Mitigations are:
[image]

How to fix the issue
Add the following directive to HTTP header.

Strict-Transport-Security: max-age=31536000; includeSubDomains

Here is the explanation of these directives:

max-age=<expire-time>

The time, in seconds, that the browser should remember that a site is only to be accessed using HTTPS.

includeSubDomains (optional)

If this optional parameter is specified, this rule applies to all of the site's subdomains as well.

Get more info about this HTTP header in [MDN Web Docs](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Strict-Transport-Security).

[image]

How to fix the issue
To enable CSP, you need to configure your web server to return the Content-Security-Policy HTTP header. Here are examples:

Example 1

A web site administrator wants all content to come from the site's own origin (this excludes subdomains.)

Content-Security-Policy: default-src 'self'

Example 2

A web site administrator wants to allow content from a trusted domain and all its subdomains (it doesn't have to be the same domain that the CSP is set on.)

Content-Security-Policy: default-src 'self' trusted.com *.trusted.com

Alternatively, the <meta> element can be used to configure a policy, for example:
<meta http-equiv="Content-Security-Policy" content="default-src 'self'; img-src https://*; child-src 'none';">

Get more info about implementing Content Security Policy for different cases in [MDN Web Docs](https://developer.mozilla.org/en-US/docs/Web/HTTP/CSP).

chuckwagoncomputing, 2022-12-29:
I wouldn't call this important, but it shouldn't hurt anything.
One gotcha could happen if our CSP policy doesn't allow images from other sources, and someone tries to add one. I think we are trying to host all wiki images ourselves, so that shouldn't happen, but I guess it's worth mentioning.

rusefillc, 2023-01-28:
I have no idea what exactly to click where and I have no priority to research :( Sorry.

mi-hol, 2023-03-26:
> I think we are trying to host all wiki images ourselves, so that shouldn't happen, but I guess it's worth mentioning.

@chuckwagoncomputing via my PRs to obsolete IMGUR as image hoster this assumption is now true.
Could a check in lint.sh be added easily to prevent other image sources?

closed 2023-05-30 by mi-hol