rusefi/rusefi_documentation #333: consolidate directory structure

mi-hol, 2023-01-02:
there are a number of redundant directories  that confuse users, therefore I'd suggest a consolidation as outlined below 

- [Special/Tools](https://github.com/mi-hol/rusefi_documentation/tree/master/Special/Tools) should be merge into [git_scripts](https://github.com/mi-hol/rusefi_documentation/tree/master/git_scripts)
- [installations](https://github.com/mi-hol/rusefi_documentation/tree/master/installations) should be merge into [Vehicles](https://github.com/mi-hol/rusefi_documentation/tree/master/Vehicles)
- [rusEFI-History](https://github.com/mi-hol/rusefi_documentation/tree/master/rusEFI-History) should be merge into [Hardware](https://github.com/mi-hol/rusefi_documentation/tree/master/Hardware)
- [bin](https://github.com/mi-hol/rusefi_documentation/tree/master/bin) should be merge into [mkdocs](https://github.com/mi-hol/rusefi_documentation/tree/master/mkdocs)
- [wiki-tools](https://github.com/mi-hol/rusefi_documentation/tree/master/wiki-tools) should be merge into [mkdocs](https://github.com/mi-hol/rusefi_documentation/tree/master/mkdocs)

are the follow directories   still required?
- [pinouts-backup](https://github.com/mi-hol/rusefi_documentation/tree/master/pinouts-backup)
- [ibom-backup](https://github.com/mi-hol/rusefi_documentation/tree/master/ibom-backup)

487 orphaned md files need a "home" and reference
- most of these files seem to belong into [Vehicles](https://github.com/mi-hol/rusefi_documentation/tree/master/Vehicles)
*[1995-3000gt.md](https://github.com/mi-hol/rusefi_documentation/blob/master/1995-3000gt.md)
*[Chevrolet-Blazer-S10-V6-1998.md](https://github.com/mi-hol/rusefi_documentation/blob/master/Chevrolet-Blazer-S10-V6-1998.md) 

files that save apparently no purpose should be deleted
- [weird-file-work-in-progress.md](https://github.com/mi-hol/rusefi_documentation/blob/master/weird-file-work-in-progress.md)
- [Quo-Vadis-Hellen.md](https://github.com/mi-hol/rusefi_documentation/blob/master/Quo-Vadis-Hellen.md)

rusefillc, 2023-01-02:
Closing because too much separate is piled into one ticket

rusefillc, 2023-01-02:
two files removed

backup folders are exactly what it sound those are - those are manual backups in case if generation scripts break primary location

mi-hol, 2023-01-02:
> two files removed

guess this is re " files that save apparently no purpose" the 2 file mentioned?

rusefillc, 2023-01-02:
I mention ticket id's in commit message thus commit above gives all exact details

chuckwagoncomputing, 2023-01-06:
What's the value add of this?
The downside is broken links.

Furthermore, if by "487 orphaned md files need a "home" and reference" you mean moving .md files into a directory, that is a WONTFIX. We are keeping flat structure for better wiki compatibility.

closed 2023-03-26 by mi-hol