rusefi/rusefi_documentation #342: Sync Wiki issue?

rusefillc, 2023-01-03:
[image]

https://github.com/rusefi/rusefi/actions/runs/3830016361/jobs/6517357463

rusefillc, 2023-01-03:
@chuckwagoncomputing do you understand root cause of issue?

chuckwagoncomputing, 2023-01-03:
It seems this comes from https://github.com/rusefi/rusefi/pull/2355
Combined with breaking sync-wiki.yaml into two files.

chuckwagoncomputing, 2023-01-03:
I'm at work now and can't test changes, but we should probably do the same thing to sync-wiki2.yaml and sync-wiki3.yaml that we are currently trying to do to sync-wiki.yaml in https://github.com/rusefi/rusefi/blob/master/.github/workflows/sync-wiki.yaml

rusefillc, 2023-01-03:
sync-wiki.yaml reset/checkout magic makes no sense to me. While each line is documented I have no idea what's the big picture why.

chuckwagoncomputing, 2023-01-03:
TL;DR bad magic github security policy that causes issues when someone commits a changed workflow file to rusefi/rusefi.wiki.git
As a workaround we undo any changes to workflow files.

closed 2023-01-03 by rusefillc