rusefi/rusefi_documentation #363: Proposal: to avoid "back and forth" "close issue" should be done only by person that opened issue

mi-hol, 2023-01-08:
There is the old saying:

**Only the customer defines a job well done**

I've noticed on several occasions that issues had been only partially fixes but the related issued was closed.

In line with quoted old saying
I'm suggestion to make the rule:

**"close issue" should be done only by person that opened issue**

Benefits:

this should avoid "back and forth" on status of issue because the person opening an issue has the time to check on success or failure of a fix. This person will have to respond within a given time (2 weeks seems adequate) otherwise issue will be automatically closed.

If we can agree on this proposal, I'll enhance the contribution guideline and will see if the process can be automated.

chuckwagoncomputing, 2023-01-08:
I think this is a bad idea, but this @rusefillc's decision to make.

rusefillc, 2023-01-08:
No, based on my experience I definitely and absolutely do not trust all customers to have such powers.

I have no resources for democracy and/or convincing everyone who can create a ticket that he has to close it.

closed 2023-01-28 by rusefillc