rusefi/rusefi_documentation #364: Sync Wiki2 - get error

mi-hol, 2023-01-08:
@chuckwagoncomputing any hint if I'm doing something wrong?

[image]

chuckwagoncomputing, 2023-01-08:
It's trying to merge your changes to rusefi/rusefi.wiki.git
In order to use the workflow for you own repo, you need to change rusefi/rusefi.wiki.git to <your-username>/rusefi.wiki.git

I've changed HOWTO Test Doc Changes to make this clearer. - https://github.com/rusefi/rusefi_documentation/pull/366

mi-hol, 2023-01-08:
I had disabled the wiki2 workflow already as you told me.

when I merge a PR, my branch is already up tp date, as the workflow reports correctly.
Why does it get started anyways, seems to be the question.

[image]

chuckwagoncomputing, 2023-01-08:
Now that's very strange. Can you provide a link to a failed run? I'm failing to find the one you posted a screenshot of.

mi-hol, 2023-01-08:
https://github.com/rusefi/rusefi/actions/runs/3866910978

this is what I get to see
[image]

chuckwagoncomputing, 2023-01-09:
OK, it's the sync wiki workflow on rusefi/rusefi - I should have seen that from the screenshot.
I'm not sure what the cause of this is; it really shouldn't be running in that case.
sync-wiki2.yaml on rusefi/rusefi_documentation has
```
git config --local user.email "action@github.com"
git config --local user.name "GitHub Action"
```
and sync-wiki.yaml on rusefi/rusefi has
```
if: ${{ github.actor != 'actions-user' }}
```
but somehow it's showing on Github that it was triggered by @mi-hol

I don't understand what's going on, but I'll dig into it some more.

chuckwagoncomputing, 2023-01-09:
I guess the merge doesn't count as a ref, so the last ref is the push. I guess I need to somehow detect if the last ref was a merge.

chuckwagoncomputing, 2023-01-10:
https://github.com/rusefi/rusefi_documentation/pull/371
and
https://github.com/rusefi/rusefi/pull/4962

should fix the wiki sync in both directions

chuckwagoncomputing, 2023-01-11:
@mi-hol This should be fixed now.

mi-hol, 2023-01-12:
Perfect! Thank you so much!

closed 2023-01-12 by mi-hol