rusefi/rusefi_documentation #367: Hardware-microRusEFI-wiring.md - Could the "TBD" & related <mark> be resolved ?

mi-hol, 2023-01-08:
@rusefillc  Could the "TBD" & related <mark> be resolved ?
[image]

rusefillc, 2023-01-28:
@mi-hol please just stop poking me with action items. My preference at the moment is to manage my action plan myself.

mi-hol, 2023-01-28:
> please just stop poking me with action items.

Well I tried to be helpful for users of documentation that will have the very same question I noticed.
Building a prototype means currently use MRE as stage 1, but how would that work with obvious errors?

mi-hol, 2023-02-21:
@JRDMcLAREN would you maybe in a position to help resolving this documentation related question?

chuckwagoncomputing, 2025-12-17:
Fixed in #635