rusefi/rusefi_documentation #379: [process proposal] peer review

rusefillc, 2023-01-13:
@mi-hol please do not self-merge PRs right away, please allow someone else on the team to review and merge

https://github.com/rusefi/rusefi_documentation/pull/378 is an example of the approach which I believe we should adjust

mi-hol, 2023-01-13:
@rusefillc the existing "[Proposal] make "master" a protected branch to enforce update via PRs only " #268 is from my view the solution to establish a defined workflow including peer review that will work independent of contributor's process knowledge.

At this stage the desired workflow seems not documented, please correct me in case I got this wrong.

To activate the setting "make "master" a protected branch "
could you please grant @chuckwagoncomputing "admin" role for rusefi/rusefi_documentation  via settings/ Access ?
GitHub documentation for this topic is [here](https://docs.github.com/en/organizations/managing-user-access-to-your-organizations-repositories/managing-an-individuals-access-to-an-organization-repository)

[image]

rusefillc, 2023-01-13:
https://github.com/rusefi/rusefi_documentation master branch is now protected

closed 2023-01-13 by rusefillc