rusefi/rusefi_documentation #383: Generate pinouts on PR

chuckwagoncomputing, 2023-01-15:
We should generate pinouts whenever someone makes a PR that touches a connector mapping, and fail the workflow if there is an error.
I think the best way to do this is with a new workflow that only generates the pinout for that particular mapping, not all the pinouts.

mi-hol, 2023-01-15:
My goal is to document a connector pin-out for a custom cable adapter (=input side) to map to a MRE pin-out (= fixed Output side of this ECU!).
This means the cable adapter should not require below attributes because they are already defined in MRE pin-out
<html>
<body>

id | the ID of the pin in the firmware | TunerStudio
-- | -- | --
ts_name | the user-facing name of the pin in TunerStudio | TunerStudio

</body>
</html>

Could this be accomplished somehow?

chuckwagoncomputing, 2023-01-16:
That's not within the scope of this issue, but yes, if you don't need to generate a .ini for TunerStudio, you can leave out fields that are only required by TunerStudio.

mi-hol, 2023-01-16:
> you can leave out fields that are only required by TunerStudio.

That's what I actually tried but no result was created

> That's not within the scope of this issue

Do you want a new issue for above failure then?

mi-hol, 2023-01-16:
https://github.com/rusefi/rusefi_documentation/issues/377#issuecomment-1384565655 shows issue of current gen_upload_pinouts.sh "Missing yaml id field in info section of" in all yaml files

chuckwagoncomputing, 2023-01-17:
Closing this issue as it has gone hopelessly off topic

closed 2023-01-17 by chuckwagoncomputing

mi-hol, 2023-01-17:
My open question is "Generate pinouts on PR" already active now?

chuckwagoncomputing, 2023-01-17:
I have made a new issue for that, #386