rusefi/rusefi_documentation #385: Documentation-Workflow.md does not exist but only Documentation_Workflow in usefi/rusefi

mi-hol, 2023-01-16:
@chuckwagoncomputing  Maybe an observation related to our issue with GH and/or past conversions/split of docs into 2 repos.
Stumbled over it when I tried to improve readme.md
Will need your background knowledge to sort it.

1. 2 files with different modification dates are reported by search in repo [rusefi/rusefi](https://github.com/rusefi/rusefi)

"Documentation_Workflow" but a past change was supposed to convert all hyphens to dash in file names
and **the file "Documentation-Workflow" does not exist in doc repo**!

[image]

2. A different situation for [Documentation_Strategy](https://github.com/rusefi/rusefi/wiki/Documentation_Strategy) but search shows totally different content and clicking search result link goes to https://github.com/rusefi/rusefi/wiki/
but **"Documentation-Strategy" does exist** in doc repo !

[image]

mi-hol, 2023-01-16:
update:

2. seems "totally different content" was a wrong impression due to search repeating files but with different "pointer to found search term", but other issue persists
3. duplicate files in doc repo (difference is _ vs -)
[image]

chuckwagoncomputing, 2023-01-16:
I reckon Github still has some no-longer-existing files indexed. Github search has always been pretty poor.

I think shop_ru.md still exists to keep from breaking links from somewhere.
wall_wetting.md should be renamed wall-wetting.md
temp_pri2022_lua.txt isn't a documentation page, so I don't think underscores matter.

mi-hol, 2023-01-16:
Just to be sure I understand your reply correctly,
Are you saying "Documentation Workflow" is no longer existing?
It is currently referenced in https://github.com/rusefi/rusefi_documentation#readme as
"https://github.com/rusefi/rusefi/wiki/Documentation_Workflow"

chuckwagoncomputing, 2023-01-17:
There are a dozen different ways you could find out for yourself whether it is existing. This issue is a waste of time. If readme.md needs to be updated, either update it or create an issue stating that it needs to be updated.

closed 2023-01-17 by chuckwagoncomputing

mi-hol, 2023-01-17:
Apologize, apparently wires got crossed completely. My last attempt to get this issue clarified.

As the title "Documentation Workflow" sounded important I did not want to remove the invalid reference as I'm still learning about structure.

> There are a dozen different ways you could find out for yourself whether it is existing.

I did try all ways I'm aware of but failed.
With my lack of understanding regarding several major change historic events I figured it's more save to ask for help

> If readme.md needs to be updated, either update it or create an issue stating that it needs to be updated.

I had started to update readme.md, that is the reason I found the issue.
This was the intent when I wrote "Stumbled over it when I tried to improve readme.md" at top.

chuckwagoncomputing, 2023-01-17:
Just to clarify as I'm not sure if this is something that's confusing you:
The github.com/rusefi/rusefi.wiki.git repository that is visible to end users at https://github.com/rusefi/rusefi/wiki is just a clone of this one. If a wiki file doesn't exist in this repository, it doesn't exist at all.