rusefi/rusefi_documentation #388: EPIC: coherent description and tooling to automatically create pinouts

mi-hol, 2023-01-17:
My goal is to document a connector pin-out for a custom cable adapter (= custom input side of Benelli ECU as example) to map to a MRE pin-out (= fixed output side to this ECU!).
This means the cable adapter should not require below attributes because they are already defined in MRE pin-out
<html>
<body>

id | the ID of the pin in the firmware | TunerStudio
-- | -- | --
ts_name | the user-facing name of the pin in TunerStudio | TunerStudio

</body>
</html>

[ ] How to accomplish?

Cross reference: https://github.com/rusefi/rusefi_documentation/issues/377#issuecomment-1384565655 shows issue of current gen_upload_pinouts.sh "Missing yaml id field in info section of" in all yaml files

chuckwagoncomputing, 2023-01-17:
A couple notes:
1. Are you confusing the id in the info section with the id that each pin has? Pinouts requires the id in the info section, but not an id for each pin.
2. Most of the boards in firmware/config/boards have not had a pinouts made for them, which is why they fail in the workflow.
3. I'm not quite sure if making a "board" is the proper thing to do for an adapter just to get a pinout, but Andrey merged it, so I guess we'll allow it. :)

mi-hol, 2023-01-17:
> 1. Are you confusing the id in the info section with the id that each pin has? Pinouts requires the id in the info section, but not an id for each pin.

exactly this seems to have happened! 
If you see me as guinea pig for the "average" beginner, every hurdle I encounter, will prevent other persons from being successful too! 
Statistics indicate that for any product "issue" encounter only 1 of 10 persons affected will report to product owner about it.
The other 9 will just walk away but tell about ~ 20 personal contacts their bad experience or complain on some unrelated places on the internet.

There is a saying "Every thing has to have a proper name" in Germany.

To use above statistics & saying to our benefit, I'd suggest to change the now duplicated "id" to unique names.
Meaning: change name "id" in info section  to name "short_url" as the description suggests its used for.
Would that make sense?

chuckwagoncomputing, 2023-01-17:
It seems your problem with Benelli_Walbro.yaml is quotes within fields - in particular `Injection Relay "Output (low)" ??`
It's technically valid yaml, but yq is spitting out invalid JSON from it.

chuckwagoncomputing, 2023-01-17:
Looks like this bug has been fixed in newer versions of YQ. Don't bother changing your yaml, I'll get us on a newer version.
We use a fixed version because the command line usage has changed on us in the past, and indeed has changed again for the version I'll be updating to.

chuckwagoncomputing, 2023-01-17:
I take that back - it wasn't a bug in yq, it was backslashes getting swallowed.

chuckwagoncomputing, 2023-01-17:
https://github.com/rusefi/rusefi/pull/4977 should solve that issue

mi-hol, 2023-02-06:
duplicate usage of generic name "id" leading to errors "by design" is not solved

mi-hol, 2023-02-13:
open:

[ ] update "how to" document

chuckwagoncomputing, 2023-02-13:
#415

mi-hol, 2023-03-26:
@chuckwagoncomputing I wasn't aware that the code for interactive-pinout lives in its own repo until recently.
Should I open an issue for "to avoid incorrect docs replace "static" views (generated by manual filtering and "copy & paste" into md files) with "dynamic" filters/sort a la Excel" there?

chuckwagoncomputing, 2023-03-26:
I don't understand what you mean by that, but go ahead.

closed 2023-05-29 by mi-hol