rusefi/rusefi_documentation #389: improve performance of link validation in md files

mi-hol, 2023-01-17:
All md files in this repository contain over 3500 links.
Validation of a single link with current shell script ./wiki-tools/brokenlinks.sh requires about 3.5 seconds on my I5 Gen2 laptop in WSL Ubuntu 22.04 environment (see example run from a slightly modified version (added timestamp display) of this script below).

Total elapsed time for a complete verification run for all md files  is therefore over 3.5 hours.

Conclusion: 
Not really feasible to run it therefore on personal laptop

Question:

- Is it run as part of doc build process?
- In case of detected issues, is the output related for someone to take corrective actions?

```bash
mholzer@PB6460b-DE:/mnt/c/Users/mholzer/Documents/git/rusefi_documentation$ ./wiki-tools/brokenlinks.sh
+ export -f escape
+ export -f escapeReplace
+ export -f searchfile
++ date
+ printf '%s\n' 'Tue Jan 17 12:35:33 CET 2023'
Tue Jan 17 12:35:33 CET 2023
+ find . -iname '*.md' -exec bash -c 'searchfile "$0"' '{}' ';'
Tue Jan 17 12:35:35 CET 2023
Tue Jan 17 12:35:39 CET 2023
Tue Jan 17 12:35:42 CET 2023
Tue Jan 17 12:35:46 CET 2023
Tue Jan 17 12:35:49 CET 2023
Tue Jan 17 12:35:52 CET 2023
Tue Jan 17 12:35:56 CET 2023
...
Tue Jan 17 12:43:03 CET 2023
Tue Jan 17 12:43:07 CET 2023
Tue Jan 17 12:43:11 CET 2023
Tue Jan 17 12:43:14 CET 2023
^C
```

chuckwagoncomputing, 2023-01-17:
On my laptop with i7-8550U, SATA SSD, and 16GB of RAM, it takes 1 minute 8 seconds. I previously said it took less than a minute, but I guess my internal clock is a little off. :)

chuckwagoncomputing, 2023-01-19:
@mi-hol I have done some magic that I suspect won't actually help you much.
Try https://github.com/chuckwagoncomputing/rusefi_documentation/blob/26497419162c9f179ecb4c0e67d5040a335f06af/wiki-tools/brokenlinks.sh
It will print the time of every link checked, so we know how long each takes.

On my laptop it now takes about 15 seconds.

chuckwagoncomputing, 2023-01-19:
Further, you can try
https://github.com/chuckwagoncomputing/rusefi_documentation/blob/1b72ad1e2ac0d612b547c72b152575411cce65b5/wiki-tools/brokenlinks.sh
Which cuts the time down to 11 seconds on my laptop.

chuckwagoncomputing, 2023-01-19:
For the fun of it, I tried on my AMD 3400G desktop, and got ~7.3 seconds. I tried copying it to a PCIE gen 3 NVME SSD, and it only improved slightly to ~7.15.
I need to build a faster computer to run my bash scripts faster.

mi-hol, 2023-01-19:
slow performance is likely caused WSL2 being a VM and my old I5 Gen 2 processor

Validating OEM-connectors.md requires 49s

 ./wiki-tools/brokenlinks_test.sh OEM-connectors.md
11:46:38.783477062
11:46:38.794936063
11:46:38.804095863
11:46:38.814390964
11:46:38.825034565
11:46:38.835364066
11:46:38.847255666
11:46:38.857783267
11:46:38.867884568
11:46:38.878546469
11:46:38.888932870
11:46:38.899763170
11:46:38.910647871
11:46:38.922003372
11:46:38.932323973
11:46:38.944181374
11:46:40.772543110
.....
11:47:23.932571341
11:47:25.516969406
11:47:25.529392108
11:47:25.541780210
11:47:25.556097512
11:47:25.570733315
11:47:25.585222417
11:47:27.369675815
11:47:27.382045717
11:47:27.394159819

chuckwagoncomputing, 2023-01-20:
I've done as much as I can with brokenlinks.sh.
If you want to continue further with other solutions, I will leave this open.

mi-hol, 2023-01-20:
mholzer@PB6460b-DE:/mnt/c/Users/mholzer/Documents/git/rusefi_documentation$ ./wiki-tools/brokenlinks.sh
(34secs)
mholzer@PB6460b-DE:/mnt/c/Users/mholzer/Documents/git/rusefi_documentation$ ./wiki-tools/brokenlinks.sh *.md
  (3secs)
above commands finish in less than 1 minute now, but what did they do?
Not indication given

chuckwagoncomputing, 2023-01-20:
> mholzer@PB6460b-DE:/mnt/c/Users/mholzer/Documents/git/rusefi_documentation$ ./wiki-tools/brokenlinks.sh *.md
(3secs)

If that finished in 3 seconds there is almost certainly a bug. I'll take a look when I get home from work today.

chuckwagoncomputing, 2023-01-23:
_Sidebar.md and _Footer.md are specific to Github Wiki, and due to the brokenness of Github Wiki, contain links that are "broken" if interpreted within the context of Wiki3 (or even other pages on Wiki2!)

brokenlinks.sh knows to exclude these files if you run it without passing it any files, but if you want to pass a list of all .md files to it, you will have to exclude those. There's a way to do that in bash, I've almost never used it, and don't remember how, but the relevant doc is https://www.gnu.org/savannah-checkouts/gnu/bash/manual/bash.html#Pattern-Matching

mi-hol, 2023-04-03:
@chuckwagoncomputing I'd like to prevent image URL links to "https://user-images.githubusercontent.com/" and IMGUR hoster.

Is the correct place to add this code to be added in brokenlinks.sh function checkurl or is there a better solution?

rusefillc, 2023-04-03:
@chuckwagoncomputing I'd like to keep image URL links to "https://user-images.githubusercontent.com/"

chuckwagoncomputing, 2023-04-03:
Yep, that's a good place. The check for HTTP links is on [line 30](https://github.com/rusefi/rusefi_documentation/blob/e934c888bd2eb738f0247f7a5ac425dd6b1c5f98/wiki-tools/brokenlinks.sh#L30).

mi-hol, 2023-04-03:
> @chuckwagoncomputing I'd like to keep image URL links to "https://user-images.githubusercontent.com/"

@rusefillc keeping this will have negative impacts for:
https://github.com/rusefi/rusefi_documentation/issues/459
https://github.com/rusefi/rusefi_documentation/issues/1
https://github.com/rusefi/rusefi_documentation/issues/431

in the long run as the repo is no longer in control of the full content used!

This has a very big impact therefore and should not be decided lightly!

rusefillc, 2023-04-03:
@mi-hol so we have a case of "humans want one things, machine wants something else". Meaning that more automation is needed to automatically convert one-paste-click images from user-images into Images/

mi-hol, 2023-04-04:
@rusefillc would you please let me know how automation could create a meaningful and correct name from the currently available information like below, I'd be happy to implement it.
this the available info for automation:
`![image](https://user-images.githubusercontent.com/17059674/177897778-0f450933-71eb-48d5-97cb-81741f11061e.png)`

rusefillc, 2023-04-04:
@mi-hol might be saying that with extra effort we get readable names if we go fully manual image add

I am saying that better image names are not always worth the effort

While we can have automation to suck such github clipboard images into repo I am not sure why

closed 2023-05-30 by mi-hol