rusefi/rusefi_documentation #395: First image users are supposed to see is broken

rusefillc, 2023-01-19:
open https://github.com/rusefi/rusefi
click "wiki"

actual result:
[image]

looks like images drama is still rolling strong :(

chuckwagoncomputing, 2023-01-19:
This seems to be browser-dependent. It works perfectly for me in Firefox.
In Chrome-based browsers, it doesn't work on initial load, but if you reload the page, then it works.
Also, it only fails when clicking the tab from `https://github.com/rusefi/rusefi`. If you are on a different "tab", it works properly. If you click the link from any other page, even on Github, it works properly.
When it doesn't work, it's trying to load
`https://github.com/rusefi/wiki/Images/Latest_bundle_border.png` instead of
`https://github.com/rusefi/rusefi/wiki/Images/Latest_bundle_border.png`

The href in the link doesn't change when reloading the page.

I wonder if Chrome is somehow starting to fetching the image before it changes the "current url".

chuckwagoncomputing, 2023-01-20:
I have filed a Chromium bug:
https://bugs.chromium.org/p/chromium/issues/detail?id=1408945

chuckwagoncomputing, 2023-01-20:
According to the response from a Chromium dev, this appears to be an issue with Github and/or a Javascript library they are using. I have filed a report with Github.

rusefillc, 2023-01-21:
Wow impressive response time from chromium guys!

chuckwagoncomputing, 2023-01-23:
Github response
>I do see this problem in multiple browsers.
Thank you for providing a link to the Google bug report for this, I expect that their analysis will point us in the right direction to correct this. I have forwarded your report to the team for investigation. We'll keep you updated on the status of this report.

chuckwagoncomputing, 2024-03-09:
This seems to be fixed

closed 2024-03-09 by chuckwagoncomputing