rusefi/rusefi_documentation #406: make link to rusEFI Store consistent

mi-hol, 2023-01-31:
Currently several URL targets are used, leading to confusion

```
[rusEFI store](https://www.shop.rusefi.com/shop) seems the only correct one
```
incorrect seem to be:
```
[<https://www.tindie.com/stores/russian/> our store]

[Ebay Store](https://www.ebay.com/itm/333532119947)
[eBay Store](http://www.ebay.com/usr/rusefi)
```

To fix the links I need above findings confirmed

mi-hol, 2023-01-31:
> https://www.tindie.com/stores/russian/ yes totally incorrect

would recommend a cleanup there
[image]

mi-hol, 2023-02-02:
as it contradicts with #259 and creates a very inconsistent "shopping" experience regarding prices
[image]

rusefillc, 2023-02-02:
retired all duplicates https://www.tindie.com/stores/russian/

closed 2023-03-07 by mi-hol