rusefi/rusefi_documentation #411: requirement to manually sync navigation bar on wiki3 und wiki2 results in inconsistencies

mi-hol, 2023-02-02:
Could the manual process be replaced with automation to avoid this source of error leading to inconsistencies?

i.e. use wiki3 [mkdocs.yaml](https://github.com/rusefi/rusefi_documentation/blob/master/mkdocs/mkdocs.yaml) 
  to create wiki2 [_Sidebar.md](https://github.com/rusefi/rusefi_documentation/blob/master/_Sidebar.md)

chuckwagoncomputing, 2023-02-02:
Maybe, but it would be a bit of a mess.
You'd have a workflow that committed back to the repo, which would trigger the sync-wiki3 workflow again.
I don't think that would really be a problem, but it's an inelegant solution to a small problem.

mi-hol, 2023-03-09:
manual edit in Wiki3 required for "boards" section to match wiki2
[image]

mi-hol, 2023-03-09:
address via https://github.com/rusefi/rusefi_documentation/pull/439

mi-hol, 2023-03-09:
@rusefillc navigation on Wiki3 misses some Plug&Play boards, should they be added?

wiki2:
[image]

wiki3:
[image]

mi-hol, 2023-03-09:
well I'm only interested to fix a content inconsistency.
Currently  in Wiki3 (showing 2 Miata models) vs Wiki2 (showing 4 Miata models).
Wiki2 is usually considered the "master" therefore the 2 missing models should be added to Wiki3.

mi-hol, 2023-06-13:
removing myself as my hands are tied without support

closed 2023-08-19 by mi-hol