rusefi/rusefi_documentation #418: discuss,  agree and write down common norms of following up or giving up

mi-hol, 2023-02-12:
As we had several cases of stuck conversation Andrey suggested this subject

I'll create a draft based on others teams norms and will ask for feedback

rusefillc, 2023-02-12:
Realistically we should automate first, it should be machines enforcing rules

chuckwagoncomputing, 2023-02-12:
https://github.com/actions/stale is available.
It can be configured so that we can manually flag an issue or PR as stale (or whatever label we want, e.g. "needs-followup"), and will close the issue after n days since being labeled.
By default it will automatically label issues/PRs as well, but I don't think we want that.

rusefillc, 2023-02-12:
Nice! Let me do something with chibios PRs so that I am not the worst offender

rusefillc, 2023-02-14:
I've cleaned up https://github.com/rusefi/rusefi/pulls to my liking

@chuckwagoncomputing would you be available to start enabling https://github.com/actions/stale? Something like a 14, 30 and 60 poke with a comment to begin with maybe?

chuckwagoncomputing, 2023-02-14:
@mi-hol has assigned this to himself

mi-hol, 2023-02-14:
> @mi-hol has assigned this to himself

I've only assigned to full user story as EPIC to me.
This doesn't mean I'll own all agreed actions!

@rusefi/doc-team 
From above comments I'd understand that a first action is to implement a workflow that implements "reaction to aging of issues".
I'll create a separate issue and add link to it in first post.

mi-hol, 2023-02-14:
based on[ this  feedback](https://github.com/rusefi/rusefi_documentation/issues/419#issuecomment-1430103802) it seems that
- more labels are required to capture state of an issue
- a state transition diagram should show how labels are used to trigger actions

mi-hol, 2023-02-14:
|Label|intention|
|-|-|
| bug | Something isn't working |
|content-is-needed ||
|duplicate|This issue or pull request already exists|
|enhancement|New feature or request|
|feedback requested|requestor or contributor needs to provide feedback|
|good first issue|Good for newcomers|
|help wanted|additional support is needed|
|IMPORTANT ||
|invalid|This doesn't seem right|
|low prio ||
|low-hanging-fruit||
|not planned||
|question|Further information is requested|
|waiting-on-feedback|blocked task, requestor or contributor needs to provide feedback|
|wiki location & process change| summarize topic|
|wiki search optimization| summarize topic|
|wontfix|This will not be worked on|

nmschulte, 2023-02-14:
This is an idea outside the immediate scope of this here goal, but it seems useful if this process can add a simple checklist of "possible concerns" for PRs, perhaps with some intelligence behind it:
- are changelog entries added (or necessary)?
- are changes to to rusef_documentation accounted for (or necessary)?
- are firmware config version bumps necessary?

mi-hol, 2023-02-15:
A team with extreme high number of issues, contributors and changes is the VSCODE team.
They invested heavily in [process documentation](https://github.com/microsoft/vscode/blob/main/CONTRIBUTING.md) and supporting automation.
Of specific relevant is their [issue tracking](https://github.com/microsoft/vscode/wiki/Issue-Tracking) workflow and the supporting https://github.com/microsoft/vscode/blob/main/CONTRIBUTING.md#automated-issue-management

Critical for them is the rule:

* Automatically closes any issue labeled `needs-more-info` if there has been no response in the past 7 days.

mi-hol, 2023-05-30:
As no comments have been received and no "team" seems to exist, I'll close this issue

closed 2023-05-30 by mi-hol