rusefi/rusefi_documentation #419: implement a workflow that implements "automatic aging of issues and pull requests".

mi-hol, 2023-02-14:
https://github.com/actions/stale is available to start with.

Specification of intended behavior:

- assign label "staled" to  issue or PR after 30 days and add comment explaining next action and timeline
- will close all items labeled  with "staled"   after 60 days since being labeled.

This means: 
- items can be exempted from aging by removing the label manually!
- just adding a comment will NOT clear the label "staled"

chuckwagoncomputing, 2023-02-14:
This is very different to what @rusefillc specified [here](https://github.com/rusefi/rusefi_documentation/issues/418#issuecomment-1429767252).

mi-hol, 2023-02-14:
I know but:
1. the requirement was extremely vague 
2. from experience I know that workflow require labels to work correctly
I've added the label "feedback requested" to give others a chance to comment and have added from your feedback "not planned" to signal that it's not ready for implementation yet.

closed 2023-08-19 by mi-hol