rusefi/rusefi_documentation #421: should "rusefi/firmware/tunerstudio/docs/readme.MD" be moved to doc repo?

mi-hol, 2023-02-19:
Several questions related to TunerStudio docs

1. should "rusefi/firmware/tunerstudio/docs/readme.MD" be moved to doc repo?
2. current reference "[image]" does not draw correctly. Reason unclear, but maybe file name containing  blanks & special chars
3. "Tuner Studio dialogs as of May 2017" could the pictures be refreshed and file names without blanks & special chars be used?

rusefillc, 2023-02-19:
so much stuff piles together :(

please create individual tickets with https://xyproblem.info/
I have no motivation to move files since I see no explanation of why such actions would benefit who

mi-hol, 2023-02-19:
Fact:
there is a pile of obsolete and unmaintained docs scattered over many places

Problem:
Users are confused on state of rusEFI and its capabilities. Leading to potential new user to choose instead another solution.
Cleanup activities are blocked instead of getting supported (see https://github.com/rusefi/rusefi_documentation/pull/404)

rusefillc, 2023-02-19:
Fact: i ask you be to less vague and you go more vague :)

rusefillc, 2023-02-19:
See https://github.com/rusefi/rusefi/commit/946b31296a6abb899ddaa9fee21951f5af015400 for my thoughts on this

PS: @mi-hol in my opinion we are facing negative consequence of your choice to make a shortcut and put more stress on me.

In my opinion higher quality ticket would have explicit https://github.com/rusefi/rusefi/tree/master/firmware/tunerstudio/docs URL

mi-hol, 2023-02-21:
blocked until https://github.com/rusefi/rusefi_documentation/issues/423 is closed

mi-hol, 2023-05-29:
not feasible

closed 2023-05-29 by mi-hol