rusefi/rusefi_documentation #422: trigger page should be converted from numeric file names to full file names

rusefillc, 2023-02-19:
https://raw.githubusercontent.com/rusefi/rusefi_documentation/master/All-Supported-Triggers.md has both legacy ``Images/triggers/trigger_49.png`` and newer ``Images/triggers/trigger_TT_SUBARU_EZ30.png`` referecences

we shall migrate all references to full file names and remove all files with numbers

https://github.com/rusefi/rusefi/blob/master/firmware/controllers/algo/engine_types.h#L309

rusefillc, 2023-02-19:
@mi-hol can you please help?

mi-hol, 2023-02-19:
@rusefillc in case we can resolve #421 with an agreement, yes

rusefillc, 2023-02-19:
Grrrrrrrrrrrr :(

mi-hol, 2023-02-21:
@rusefillc should the same picture be referenced more than once?
Current example:
[image]

PR #426 (2023-02-24): fix #422
closes #422
commit d6e1e276 (2023-02-24): fix #422 (#426)

* remove 3 years old task list

* fix '422

* fix trigger_55.png
closed 2023-02-24 by rusefillc