rusefi/rusefi_documentation #423: grant mi-hol read access to rusEFI forum to enable doc URL link checks

mi-hol, 2023-02-20:


mi-hol, 2023-03-06:
@rusefillc may I kindly remind you of this blocking task to grant me read access to rusEFI forum?

rusefillc, 2023-03-06:
Yes it's on the list :( My bad

mi-hol, 2023-03-19:
this is blocking research for https://github.com/rusefi/rusefi_documentation/issues/255 too

rusefillc, 2023-04-07:
@mi-hol status update: I do not see a trivial way to add a read-only user. something is off, at the moment it looks like I only have one user for the database that's odd but that's how it looks like

mi-hol, 2023-04-07:
if you'd tell me the DBMS used for forum I'd be happy to provide the SQL statements to add a read-only group and add user(s) to it

rusefillc, 2023-04-07:
phpbb 3.2.11
MySQL 5.7

mi-hol, 2023-04-18:
tried to get support from Andrey (who is the only person with access to forum database) via SLACK messages, but no progress since 13Apr2023

closed 2023-07-08 by rusefillc