rusefi/rusefi_documentation #431: BUILD: png images could be reduced in size by avg of 15% using oxipng

mi-hol, 2023-03-06:
oxipng available from[ shssoichiro/oxipng](https://github.com/shssoichiro/oxipng#installing)
would reduce size of current png images on avg of 15% and speed up loading of wiki pages.

@chuckwagoncomputing Does it make sense to add  oxipng to build task?

Test results achieved via recommended options `"C:\Program Files\oxipng\oxipng.exe" -o 4 -i 1 --strip safe *.png`

original size of Images directory:

[image]

reduced size of Images directory:

[image]

chuckwagoncomputing, 2023-03-07:
If it's really lossless, it makes more sense to run it once.

mi-hol, 2023-03-13:
with running only once, updates & additions wont benefit. 
Maybe instead of "build" I should write on PR validation?

closed 2023-07-08 by rusefillc