rusefi/rusefi_documentation #440: linter to be invoked in connection with wiki2 direct edits

rusefillc, 2023-03-09:
direct https://github.com/rusefi/rusefi/wiki/ edits are an amazing tool for myself @rusefillc 

Linter is an amazing tool helping wiki quality

We need to bring these two closer together. Can we have lint invoked during syncs or periodically or else?

rusefillc, 2023-03-09:
Actually weird, I clearly remember sync failing when we had errors in miata file https://github.com/rusefi/rusefi_documentation/commit/9687323aaa2f0c613fe6875c2d59ac2bd1cd4159

So it look like lint is invoked at least for some sync, could it be invoked only for one direction and not for both?

rusefillc, 2023-03-09:
@chuckwagoncomputing please help :)

chuckwagoncomputing, 2023-03-09:
Re: sync failing because of miata file:
There are two different checks:
1. MKDocs' built-in checks
2. The Lint job

The failed sync was because of 1.
2 doesn't happen with the wiki sync job.

Re: Adding lint check to wiki sync
What would be the expected behavior of a failed lint?
1. Failing wiki sync is considered harmful because it can cause repository states that are no longer auto-mergeable.
2. Having the linter run after the sync would work, and you would receive an email notifying you of the failure.
However, the repository state would be "broken", and if someone else tried to make a PR before you fixed what you've broken, the linter job on the PR would fail. This is kind of how things already are, so it would be an improvement for you to get an email telling you that you did something wrong.

rusefillc, 2023-03-09:
> Having the linter run after the sync would work, and you would receive an email notifying

This would be best balance of drama and safety!

rusefillc, 2023-03-09:
> This is kind of how things already are, so it would be an improvement for you to get an email telling you that you did something wrong.

At the moment I do not get email thus do not notice breakage

PR #444 (2023-03-11): Also lint on push to master
This should fix #440
closed 2023-03-11 by rusefillc