rusefi/rusefi_documentation #479: nightly build for WIKI with full validation enabled to catch errors early

mi-hol, 2023-04-04:
Currently only changed documents are rebuild and validated.
Some changes like bulk-updates of links including deletes of obsolete file go unnoticed for their unintentional side effects until a user complains or a change in an affected file occured.

Proposal:

- a nightly WIKI rebuild with all defined validation enabled to a temporary location 
- the output is only copied to production WIKI site when no errors are found

mi-hol, 2023-04-04:
my PR https://github.com/rusefi/rusefi_documentation/pull/426 deleted files in commit https://github.com/rusefi/rusefi_documentation/pull/426/commits/593697fe7346ed5a6675511e49d13cc2abc555ae that had negative side effects for files fixed by Andrey's commit https://github.com/rusefi/rusefi_documentation/commit/9687323aaa2f0c613fe6875c2d59ac2bd1cd4159

chuckwagoncomputing, 2023-04-04:
Oh I see now that it didn't actually run a full check if files were deleted, only if they were renamed or moved.
https://github.com/rusefi/rusefi_documentation/pull/481

mi-hol, 2023-05-27:
> * build to a temporary location
> * the output is only copied from temporary location to production WIKI site when no errors are found

is there any chance this could happen?

chuckwagoncomputing, 2023-05-29:
Why is requiring lint to pass before merging a PR not sufficient?

rusefillc, 2023-05-29:
> Why is requiring lint to pass before merging a PR not sufficient?

In reality I am definitely guilty of pushing outside of PR process, and so far I am planning to keep editing directly :( Not trying to steer anything just stating the facts.

mi-hol, 2023-05-29:
makes not sense

closed 2023-05-29 by mi-hol