rusefi/rusefi_documentation #480: revisit lint settings

rusefillc, 2023-04-04:
https://github.com/rusefi/rusefi_documentation/actions/runs/4608587957/jobs/8144555821

```
OEM-Docs/Honda/E24-SEFMJ-white-civic-si/readme.md:1 MD041/first-line-heading/first-line-h1 First line in a file should be a top-level heading [Context: "See RandomToolHondaKPacketAnal..."]
OEM-Docs/Honda/E24-SEFMJ-white-civic-si/readme.md:1:[5](https://github.com/rusefi/rusefi_documentation/actions/runs/4608587957/jobs/8144555821#step:4:6)9 MD047/single-trailing-newline Files should end with a single newline character
```

Do we see value in both of these errors? I can maybe see some merit in MD041 but MD047 why do we torture ourselves?

rusefillc, 2023-04-04:
https://github.com/rusefi/rusefi_documentation/commit/c45ae2da12a0ab0434154bc915a3df2b24eb8ee9

rusefillc, 2023-04-05:
another strange error https://github.com/rusefi/rusefi_documentation/pull/482/commits/2f7a601519c0793d37e3e8e2fda7673575a5b61e

rusefillc, 2023-04-05:
> What's strange about that? It's telling you what's wrong.

Why do we case? What is the value of those rules?

chuckwagoncomputing, 2023-04-05:
#308
Linting helps us be sure that we are writing standard Markdown, which is most likely to be rendered correctly in all our outputs.
- Github wiki
- MkDocs wiki
- PDFs

rusefillc, 2023-04-05:
@chuckwagoncomputing I absolutely support lint in general

What I am asking is
* is it an "all or nothing" deal
* if we have flexibility I question if we need https://github.com/DavidAnson/markdownlint/blob/main/doc/Rules.md#md047---files-should-end-with-a-single-newline-character

chuckwagoncomputing, 2023-04-05:
We can disable rules. I haven't tested, so I don't know if any of our "outputs" care about a newline at the end of the file.
What editor did you use to create that file? Normally editors save files with a newline at the end.

mi-hol, 2023-04-09:
Another option would be to automatically fix all fixable rule violations.
markdownlint-cli2 has this option and I tried it successfully already.

mi-hol, 2023-05-15:
@rusefillc shall "automatically fix all fixable rule violations" be enabled?
in case not, please just close this issue

mi-hol, 2023-05-25:
As Andrey did not respond, I'd believe it's no longer an issue or he lost interest.
In either case I'm inclined to just close it.

closed 2023-05-29 by chuckwagoncomputing

rusefillc, 2023-06-19:
https://github.com/updownpress/markdown-lint/blob/master/rules/040-fenced-code-language.md is a weird one

mi-hol, 2023-06-19:
> fenced-code-language is a weird one

I love it as it makes programming language used perfectly visible but without proper tooling its a pain.
Therefore I always use the markdownlint extension in my IDE

rusefillc, 2023-06-20:
https://github.com/rusefi/rusefi_documentation/actions/runs/5316170214/jobs/9625402565

I made it crash

[image]

mi-hol, 2023-06-20:
no surprise because its getting abused.
These files (https://github.com/rusefi/rusefi_documentation/tree/master/wip) should never show up in documentation as md files!