rusefi/rusefi_documentation #493: kia EMS_DCT11 packet checksum

rusefillc, 2023-07-08:
I have a 2013 Kia Rio and i want to have gauge cluster happy

https://github.com/rusefi/opendbc/blob/008104f9400b6161461db40d66ce13b34ea9ce79/hyundai_kia_generic.dbc#L487C1-L497C58 says last byte is some sort of a checksum? 

One odd thing: for same payload I have three different values of last byte

```
    291       176.930 DT     0080 Rx 8  00 17 7A 0F 1B 2C 1B *C6*
    309       186.908 DT     0080 Rx 8  00 17 7A 0F 1B 2C 1B *B7*
```

Does this checksum account for something else other than the payload?

```
      7         7.802 DT     0080 Rx 8  00 17 06 0F 1F 2C 1F 05 
     24        17.391 DT     0080 Rx 8  00 17 1C 0F 1F 2C 1E 96 
     39        26.793 DT     0080 Rx 8  00 17 2A 0F 1F 2C 1E 97 
     59        37.376 DT     0080 Rx 8  00 17 2A 0F 1D 2C 1E A8 
     74        47.630 DT     0080 Rx 8  00 17 38 0F 1D 2C 1E A9 
     91        56.945 DT     0080 Rx 8  00 17 42 0F 1D 2C 1E EA 
    108        67.375 DT     0080 Rx 8  00 17 42 0F 1D 2C 1E DB 
    125        77.238 DT     0080 Rx 8  00 17 62 0F 1C 2C 1C DC 
    141        87.206 DT     0080 Rx 8  00 17 62 0F 1C 2C 1C CD 
    160        97.437 DT     0080 Rx 8  00 17 62 0F 1C 2C 1C BE 
    174       106.834 DT     0080 Rx 8  00 17 56 0F 1C 2C 1C 7F 
    191       116.830 DT     0080 Rx 8  00 17 56 0F 1C 2C 1C 60 
    207       127.232 DT     0080 Rx 8  00 17 56 0F 1C 2C 1C 51 
    223       136.931 DT     0080 Rx 8  00 17 6A 0F 1B 2C 1B 12 
    240       146.772 DT     0080 Rx 8  00 17 72 0F 1B 2C 1B 73 
    258       156.841 DT     0080 Rx 8  00 17 72 0F 1B 2C 1B 64 
    275       167.224 DT     0080 Rx 8  00 17 78 0F 1B 2C 1B F5 
    291       176.930 DT     0080 Rx 8  00 17 7A 0F 1B 2C 1B C6 
    309       186.908 DT     0080 Rx 8  00 17 7A 0F 1B 2C 1B B7 
    327       196.819 DT     0080 Rx 8  00 17 86 0F 1A 2C 1A F8 
    344       207.288 DT     0080 Rx 8  00 17 8A 0F 1A 2C 1A A9 
    361       217.188 DT     0080 Rx 8  00 17 8A 0F 1A 2C 1A 9A 
    375       226.773 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1A 7B 
    394       236.809 DT     0080 Rx 8  00 17 76 0F 1A 2C 1A CC 
    410       247.204 DT     0080 Rx 8  00 17 76 0F 1A 2C 1A BD 
    427       256.921 DT     0080 Rx 8  00 17 7A 0F 1A 2C 1A 6E 
    443       266.773 DT     0080 Rx 8  00 17 7A 0F 1A 2C 1A 5F 
    460       276.773 DT     0080 Rx 8  00 17 7A 0F 1A 2C 1A 40 
    477       287.181 DT     0080 Rx 8  00 17 70 0F 1B 2C 1B B1 
    495       297.189 DT     0080 Rx 8  00 17 6E 0F 1B 2C 1B D2 
    511       306.796 DT     0080 Rx 8  00 17 6E 0F 1B 2C 1B C3 
    528       316.821 DT     0080 Rx 8  00 17 72 0F 1B 2C 1B 64 
    544       327.216 DT     0080 Rx 8  00 17 70 0F 1B 2C 1B 75 
    562       336.921 DT     0080 Rx 8  00 17 70 0F 1B 2C 1B 66 
    577       346.757 DT     0080 Rx 8  00 17 52 0F 1C 2C 1C 37 
    595       356.831 DT     0080 Rx 8  00 17 4E 0F 1C 2C 1C 78 
    612       367.215 DT     0080 Rx 8  00 17 4E 0F 1C 2C 1C 69 
    628       376.898 DT     0080 Rx 8  00 17 5C 0F 1C 2C 1C 6A 
    644       386.734 DT     0080 Rx 8  00 17 62 0F 1C 2C 1C EB 
    662       396.826 DT     0080 Rx 8  00 17 62 0F 1C 2C 1C DC 
    679       407.365 DT     0080 Rx 8  00 17 5A 0F 1C 2C 1C 5D 
    695       417.123 DT     0080 Rx 8  00 17 60 0F 1C 2C 1C DE 
    710       427.096 DT     0080 Rx 8  00 17 60 0F 1C 2C 1C CF 
    728       436.880 DT     0080 Rx 8  00 17 7A 0F 1A 2C 1B 30 
    744       447.293 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1B 01 
    761       457.026 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1B F2 
    777       466.838 DT     0080 Rx 8  00 17 70 0F 1A 2C 1B A3 
    794       476.867 DT     0080 Rx 8  00 17 6E 0F 1A 2C 1B C4 
    811       487.287 DT     0080 Rx 8  00 17 6E 0F 1A 2C 1B B5 
    829       497.276 DT     0080 Rx 8  00 17 78 0F 1A 2C 1B F6 
    844       506.913 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1B A7 
    861       516.897 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1B 98 
    877       527.300 DT     0080 Rx 8  00 17 7E 0F 1A 2C 1A 79 
    895       536.998 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1A 8A 
    911       547.078 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1A 7B 
    928       556.919 DT     0080 Rx 8  00 17 7C 0F 1A 2C 1A 6C 
[... 173 lines cut ...]
   3836      2297.111 DT     0080 Rx 8  00 17 6C 0F 1B 2C 1B 7A 
   3852      2306.890 DT     0080 Rx 8  00 17 6A 0F 1B 2C 1B 8B 
   3869      2316.875 DT     0080 Rx 8  00 17 6A 0F 1B 2C 1B 7C 
   3885      2327.263 DT     0080 Rx 8  00 17 78 0F 1B 2C 1B 7D 
   3903      2336.980 DT     0080 Rx 8  00 17 72 0F 1B 2C 1B CE 
   3918      2346.816 DT     0080 Rx 8  00 17 72 0F 1B 2C 1B BF 
   3936      2356.891 DT     0080 Rx 8  00 17 32 0F 1E 2C 1E 80 
   3953      2367.285 DT     0080 Rx 8  00 17 18 0F 1E 2C 1E 31 
   3969      2376.979 DT     0080 Rx 8  00 17 18 0F 1E 2C 1E 22 
   3985      2386.811 DT     0080 Rx 8  00 17 14 0F 1F 2C 1F 33 
   4003      2396.876 DT     0080 Rx 8  00 17 14 0F 1F 2C 1F 24 
   4020      2407.443 DT     0080 Rx 8  00 17 14 0F 1F 2C 1F 15 
   4037      2417.347 DT     0080 Rx 8  00 17 08 0F 1F 2C 1F D6 
   4051      2426.933 DT     0080 Rx 8  00 17 10 0F 1F 2C 1F 37 
   4070      2436.940 DT     0080 Rx 8  00 17 10 0F 1F 2C 1F 28 
   4086      2447.362 DT     0080 Rx 8  00 17 30 0F 1E 2C 1E 19 
   4103      2457.079 DT     0080 Rx 8  00 17 3A 0F 1E 2C 1E 6A 
   4119      2466.926 DT     0080 Rx 8  00 17 3A 0F 1E 2C 1E 5B 
   4137      2477.520 DT     0080 Rx 8  00 17 42 0F 1D 2C 1D DC 
   4153      2487.373 DT     0080 Rx 8  00 17 46 0F 1D 2C 1D 8D 
   4171      2497.340 DT     0080 Rx 8  00 17 46 0F 1D 2C 1D 7E 
   4187      2507.530 DT     0080 Rx 8  00 17 5C 0F 1C 2C 1D 0F 
   4203      2517.092 DT     0080 Rx 8  00 17 5E 0F 1C 2C 1D D0 
   4219      2527.379 DT     0080 Rx 8  00 17 5E 0F 1C 2C 1D C1 
```

rusefillc, 2023-07-08:
https://github.com/commaai/opendbc/blob/e498d012147e00301fc443663e0ee57cbfdc3c24/hyundai_kia_generic.dbc#L433

129 is very interesting!

```
      8         8.047 DT     0081 Rx 8  40 84 5F 00 00 00 00 75 
     25        17.631 DT     0081 Rx 8  40 84 5F 00 00 00 00 66 
     40        27.038 DT     0081 Rx 8  40 84 5F 00 00 00 00 57 
     60        37.622 DT     0081 Rx 8  40 84 5F 00 00 00 00 48 
     75        47.875 DT     0081 Rx 8  40 84 5F 00 00 00 00 39 
     92        57.190 DT     0081 Rx 8  40 84 5F 00 00 00 00 2A 
    109        67.614 DT     0081 Rx 8  40 84 5F 00 00 00 00 1B 
    126        77.484 DT     0081 Rx 8  40 84 5F 00 00 00 00 0C 
    142        87.447 DT     0081 Rx 8  40 84 5F 00 00 00 00 FD 
    161        97.682 DT     0081 Rx 8  40 84 5F 00 00 00 00 EE 
    175       107.079 DT     0081 Rx 8  40 84 5F 00 00 00 00 DF 
    192       117.075 DT     0081 Rx 8  40 84 5F 00 00 00 00 C0 
    208       127.471 DT     0081 Rx 8  40 84 5F 00 00 00 00 B1 
    224       137.171 DT     0081 Rx 8  40 84 5F 00 00 00 00 A2 
    241       147.011 DT     0081 Rx 8  40 84 5F 00 00 00 00 93 
    259       157.080 DT     0081 Rx 8  40 84 5F 00 00 00 00 84 

    276       167.464 DT     0081 Rx 8  40 84 5F 00 00 00 00 75 
    292       177.168 DT     0081 Rx 8  40 84 5F 00 00 00 00 66 
    310       187.147 DT     0081 Rx 8  40 84 5F 00 00 00 00 57 
    328       197.058 DT     0081 Rx 8  40 84 5F 00 00 00 00 48 
    345       207.534 DT     0081 Rx 8  40 84 5F 00 00 00 00 39 
    362       217.434 DT     0081 Rx 8  40 84 5F 00 00 00 00 2A 
    376       227.017 DT     0081 Rx 8  40 84 5F 00 00 00 00 1B 
    395       237.054 DT     0081 Rx 8  40 84 5F 00 00 00 00 0C 
    411       247.450 DT     0081 Rx 8  40 84 5F 00 00 00 00 FD 
    428       257.166 DT     0081 Rx 8  40 84 5F 00 00 00 00 EE 
    444       267.018 DT     0081 Rx 8  40 84 5F 00 00 00 00 DF 
    461       277.018 DT     0081 Rx 8  40 84 5F 00 00 00 00 C0 
    478       287.422 DT     0081 Rx 8  40 84 5F 00 00 00 00 B1 
    496       297.429 DT     0081 Rx 8  40 84 5F 00 00 00 00 A2 
    512       307.036 DT     0081 Rx 8  40 84 5F 00 00 00 00 93 
    529       317.059 DT     0081 Rx 8  40 84 5F 00 00 00 00 84 

    545       327.457 DT     0081 Rx 8  40 84 5F 00 00 00 00 75 
    563       337.161 DT     0081 Rx 8  40 84 5F 00 00 00 00 66 
    578       346.997 DT     0081 Rx 8  40 84 5F 00 00 00 00 57 
    596       357.070 DT     0081 Rx 8  40 84 5F 00 00 00 00 48 
    613       367.455 DT     0081 Rx 8  40 84 5F 00 00 00 00 39 
    629       377.143 DT     0081 Rx 8  40 84 5F 00 00 00 00 2A 
    645       386.973 DT     0081 Rx 8  40 84 5F 00 00 00 00 1B 
    663       397.071 DT     0081 Rx 8  40 84 5F 00 00 00 00 0C 
    680       407.606 DT     0081 Rx 8  40 84 5F 00 00 00 00 FD 
    696       417.368 DT     0081 Rx 8  40 84 5F 00 00 00 00 EE 
    711       427.341 DT     0081 Rx 8  40 84 5F 00 00 00 00 DF 
    729       437.119 DT     0081 Rx 8  40 84 5F 00 00 00 00 C0 
    745       447.533 DT     0081 Rx 8  40 84 5F 00 00 00 00 B1 
    762       457.265 DT     0081 Rx 8  40 84 5F 00 00 00 00 A2 
    778       467.083 DT     0081 Rx 8  40 84 5F 00 00 00 00 93 
    795       477.107 DT     0081 Rx 8  40 84 5F 00 00 00 00 84 

    812       487.533 DT     0081 Rx 8  40 84 5F 00 00 00 00 75 
    830       497.517 DT     0081 Rx 8  40 84 5F 00 00 00 00 66 
    845       507.152 DT     0081 Rx 8  40 84 5F 00 00 00 00 57 
    862       517.142 DT     0081 Rx 8  40 84 5F 00 00 00 00 48 
    878       527.539 DT     0081 Rx 8  40 84 5F 00 00 00 00 39 
    896       537.238 DT     0081 Rx 8  40 84 5F 00 00 00 00 2A 
    912       547.323 DT     0081 Rx 8  40 84 5F 00 00 00 00 1B 
    929       557.158 DT     0081 Rx 8  40 84 5F 00 00 00 00 0C 
    946       567.519 DT     0081 Rx 8  40 84 5F 00 00 00 00 FD 
    962       577.235 DT     0081 Rx 8  40 84 5F 00 00 00 00 EE 
    979       587.630 DT     0081 Rx 8  40 84 5F 00 00 00 00 DF 
    996       597.145 DT     0081 Rx 8  40 84 5F 00 00 00 00 C0 
   1013       607.588 DT     0081 Rx 8  40 84 5F 00 00 00 00 B1 
   1028       617.400 DT     0081 Rx 8  40 84 5F 00 00 00 00 A2 
[... 180 lines cut ...]
   3870      2317.119 DT     0081 Rx 8  40 84 5F 00 00 00 00 0C 
   3886      2327.503 DT     0081 Rx 8  40 84 5F 00 00 00 00 FD 
   3904      2337.224 DT     0081 Rx 8  40 84 5F 00 00 00 00 EE 
   3919      2347.061 DT     0081 Rx 8  40 84 5F 00 00 00 00 DF 
   3937      2357.136 DT     0081 Rx 8  40 84 5F 00 00 00 00 C0 
   3954      2367.524 DT     0081 Rx 8  40 84 5F 00 00 00 00 B1 
   3970      2377.224 DT     0081 Rx 8  40 84 5F 00 00 00 00 A2 
   3986      2387.056 DT     0081 Rx 8  40 84 5F 00 00 00 00 93 
   4004      2397.116 DT     0081 Rx 8  40 84 5F 00 00 00 00 84 

   4021      2407.682 DT     0081 Rx 8  40 84 5F 00 00 00 00 75 
   4038      2417.586 DT     0081 Rx 8  40 84 5F 00 00 00 00 66 
   4052      2427.173 DT     0081 Rx 8  40 84 5F 00 00 00 00 57 
   4071      2437.180 DT     0081 Rx 8  40 84 5F 00 00 00 00 48 
   4087      2447.602 DT     0081 Rx 8  40 84 5F 00 00 00 00 39 
   4104      2457.324 DT     0081 Rx 8  40 84 5F 00 00 00 00 2A 
   4120      2467.166 DT     0081 Rx 8  40 84 5F 00 00 00 00 1B 
   4138      2477.761 DT     0081 Rx 8  40 84 5F 00 00 00 00 0C 
   4154      2487.620 DT     0081 Rx 8  40 84 5F 00 00 00 00 FD 
   4172      2497.580 DT     0081 Rx 8  40 84 5F 00 00 00 00 EE 
   4188      2507.770 DT     0081 Rx 8  40 84 5F 00 00 00 00 DF 
   4204      2517.337 DT     0081 Rx 8  40 84 5F 00 00 00 00 C0 
   4220      2527.619 DT     0081 Rx 8  40 84 5F 00 00 00 00 B1 

```

esaulenka, 2023-07-10:
> hyundai_kia_generic.dbc says last byte is some sort of a checksum?

No, it stated that 4 bits in last byte is 'Alive' (i.e. just incrementing counter) and other one is a 'ChkSum'. That is exaсtly what you found (but not documented here ;) ).

rusefillc, 2023-07-10:
My Lua script at https://gist.github.com/rusefillc/26d36ced900dc54b5fdef01e43d842d3

https://github.com/commaai/panda/blob/master/board/safety/safety_hyundai.h around line 152 has C version of same

closed 2023-07-10 by rusefillc